pvlib/pvlib-python #474: Module and inverter parameters from data sheet to pvlib-python

Jawairia123, 2018-06-11:
I am trying to model a PV system using `pvlib-python`. The PV modules and the inverter model that is required is not present in any of the databases that's been used in either pvlib or SAM by NREL. These models are:

Module   = GCL-P6/72H 330 

Inverter = SG2500HV (by Sungrow)

I want to know is there any standard way or any approach to input parameters from datasheets directly? 

Thanks in advance!

cwhanse, 2018-06-11:
Briefly, yes there are standard ways but they are not coded in pvlib.

As background, a module performance model requires certain parameter values (e.g., Rshunt for the CEC model) that are not found on datasheets.  For modules, there are techniques to obtain parameters for some module performance model (e.g., De Soto, CEC) from datasheet values.  The SAM SDK includes these functions, so you may be able to use SAM to get a set of module model parameters. 

For inverters, it's a bit more difficult.  The efficiency curves are needed; these could be digitized from the figure commonly found on an inverter cutsheet. The parameters for the Sandia inverter model can then be estimated by a fitting method.  The method is not difficult (I have a spreadsheet somewhere) and I believe it is also coded in SAM.

Estimation of model parameters from data is an important gap which pvlib currently lacks.  We welcome pull requests :)

vanpnguyen, 2018-06-14:
cwhanse,

Could you provide step by step procedure to load a locally saved PV module file (CEC model) to Python?
Also could you provide the template file for the CEC model so that i can populate the parameters from datasheet / SAM SDK?

Thanks,

cwhanse, 2018-06-15:
We don't yet have the CEC model implemented but you can use the Desoto model with CEC parameters, it will give you very similar results. 

The Desoto model parameters are:

-  `I_L_ref` (light current at STC)
-  `I_o_ref` (dark current at STC)
-  `a_ref`  (product of diode ideality factor, number of series cells and cell thermal voltage at STC)
-  `R_s` (series resistance)
-  `R_sh_ref` (shunt resistance at STC)
-  `alpha_sc` (in A/K, temperature coefficient of Isc)
- 

If you have those values in your locally saved file, then you can pass them as keyword arguments to 

(['a_ref', 'I_L_ref', 'I_o_ref', 'R_sh_ref', 
305                                 'R_s', 'alpha_sc', 'EgRef', 'dEgdT'],

mikofski, 2018-06-17:
seems like in the future this issue might be better asked at [pvlib-python Google group](https://groups.google.com/forum/m/#!forum/pvlib-python) thanks!

Jawairia123, 2018-06-18:
Hi, 

I tried using Desoto model with CEC module parameters and it's working. I am looking at system losses at the moment. So issue of working on CEC database is kinda resolved.

I was able to run pvwatts model but there I found this issue that currently the `modelchain` can't handle 
(pvwattts modeled) system with multiple modules in series and parallel to an inverter. So that's a drawback. 

It doesn't end here. There are quite a lot number of module and inverter models that we can't find in Sandia/CEC databases and how we can implement them directly having only datasheet values is a big gap. 

I am still looking to possible solution and suggestions. 

Regards
Jawairia

Jawairia123, 2018-06-18:
Hi,

I tried using Desoto model with CEC module parameters and it's working. I am looking at system losses at the moment. So issue of working on CEC database is kinda resolved.

I was able to run pvwatts model but there I found this issue that currently the modelchain can't handle
(pvwattts modeled) system with multiple modules in series and parallel to an inverter. So that's a drawback.

It doesn't end here. There are quite a lot number of module and inverter models that we can't find in Sandia/CEC databases and how we can implement them directly having only datasheet values is a big gap.

I am still looking to possible solution and suggestions.

Regards
Jawairia

cwhanse, 2018-06-18:
We prefer questions about use of pvlib to appear at https://groups.google.com/forum/m/#!forum/pvlib-python

Getting to module model parameters from datasheets is not currently supported in pvlib - see my response above. You may want to look at the capability in SAM - see this thread [https://sam.nrel.gov/node/69678](url)

Best regards,

Cliff

wholmgren, 2018-06-18:
This a combination of usage question and duplicate of other issues, so closing the issue.

closed 2018-06-18 by wholmgren