pvlib/pvlib-python #476: pvlib modelchain with pvwatts model

Jawairia123, 2018-06-13:
Hi, 

I am trying to run Modelchain with pvwatt model but it seems that the `modules_per_string` and `strings_per inverter ` doesn't have any affect on the total output. 

I am not sure why is it so. 
May be ModelChain isn't supporting so. If that's the case how can I achieve the desired result?

Here is my code: 

Thanks in advance
```
# built-in python modules
import os
import inspect

# scientific python add-ons
import numpy as np
import pandas as pd

# plotting stuff
# first line makes the plots appear in the notebook
%matplotlib inline 
import matplotlib.pyplot as plt
import matplotlib as mpl
# seaborn makes your plots look better
try:
    import seaborn as sns
    sns.set(rc={"figure.figsize": (12, 6)})
    sns.set_color_codes()
except ImportError:
    print('We suggest you install seaborn using conda or pip and rerun this cell')

# finally, we import the pvlib library
import pvlib

tmy = pd.read_csv("http://re.jrc.ec.europa.eu/pvgis5/tmy.php?lat=29.74&lon=40.10")

tmy.Date = pd.to_datetime(tmy.Date, format='%Y-%d-%m %H:%M:%S')

tmy.rename(columns={' Ghor':'ghi','Dhor':'dhi','DNI':'dni','Tair':'temp_air',
                        'Ws':'wind_speed'},inplace=True)					
						
tmy.set_index(tmy['Date'],inplace=True)
    #Drop unnecessary column

tmy = tmy.drop('Date', 1)
tmy = tmy.drop('RH', 1)
tmy = tmy.drop('IR', 1)
tmy = tmy.drop(' Wd', 1)
tmy = tmy.drop('Pres', 1)

cec_modules = pvlib.pvsystem.retrieve_sam('CECMod')
cec_inverters = pvlib.pvsystem.retrieve_sam('cecinverter')

#module = cec_modules['Jinko_Solar_JKM320P_72_V']  model chain doesnt support cec 
#inv = cec_inverters['ABB__PVS980_58_2000kVA_K__N_A_V__CEC_2018_']

lat = 29.74
lon = 40.10
altitude = 676
tz = 'Etc/GMT+3'  

loc = pvlib.location.Location(latitude=lat,longitude= lon,tz=tz)

#model = pvwatts 
pvwatts_system = pvlib.pvsystem.PVSystem(module_parameters={'pdc0': 320, 'gamma_pdc': -0.0041},inverter_parameters={'pdc' : 3200000, 'pdc0' : 2024292, 'eta_inv_nom':0.988, 'eta_inv_ref':0.986},surface_tilt = 20, surface_azimuth=0,
                    modules_per_string=30,strings_per_inverter=267, albedo = 0.2)
					
mc = pvlib.modelchain.ModelChain(pvwatts_system, loc, transposition_model ="perez",aoi_model = 'ashrae',spectral_model='no_loss')
print(mc)
mc.run_model(times=tmy.index,weather=tmy)
a = mc.ac
a = pd.Series.to_frame(a)
a = a * 530  # 530 = number of inverters in the system 

a['month'] = a.index
a.month = a.month.dt.month
monthly = a.groupby('month').sum()
```

wholmgren, 2018-06-13:
https://stackoverflow.com/questions/49550656/run-pvlib-modelchain-with-pvwatts-model/50165303#50165303

The work around is to scale your ``module_parameters ['pdc0']``. Pull requests welcome for improving the functionality and/or documentation.

thunderfish24, 2018-06-21:
It seems that the system scaling provided by `PVSystem.scale_voltage_current_power()` is a system-level entity that should be included in `PVSystem.sapm()` and `PVSystem.singlediode()` computations, in addition to adding this to `PVSystem.pvwatts_dc()`. Currently, a higher level `ModelChain` function does this (except for pvwatts, as discussed above). If folks agree to this, then a question arises as to if the corresponding wrapped functions in `pvsystem.py` should still only calculate `singlediode()` for a single module/device instead of the whole system. (ATM, I think that they should.)

wholmgren, 2021-01-20:
@cwhanse we need this for SPI. Do you have any concern with adding this

```python
        self.results.dc = self.system.scale_voltage_current_power(
            self.results.dc,
            unwrap=False
        )
```

to 

https://github.com/pvlib/pvlib-python/blob/56971c614e7faea3c24013445f1bf6ffe9943305/pvlib/modelchain.py#L732-L735

?

Or do you think we should go ahead with @markcampanelli's suggestion above? I think @markcampanelli's suggestion is better on the merits but it's a much bigger change and I don't know how to do it in a way that wouldn't cause user code to return unexpected answers.

cwhanse, 2021-01-20:
I don't have a problem patching that into `pvlib.modelchain.ModelChain.pvwatts_dc`. I think it's an oversight that the scaling was left out, since it differs from the `sapm` and `singlediode` methods.

wholmgren, 2021-01-20:
I think we left it out because it's arguably a departure from the pvwatts model in which you're typically specifying the pdc0 of the entire system. But I don't see a problem with the extension within our data model.

cwhanse, 2021-01-20:
want me to open a PR? Or have you got it?

wholmgren, 2021-01-20:
Would be great if you can do it.

PR #1138 (2021-01-21): Add scale_voltage_current_power to ModelChain.pvwatts_dc
 - [x] Closes #476 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the G [... cut]
closed 2021-01-21 by wholmgren