pvlib/pvlib-python #477: NaN returned from solar_azimuth_analytical

Sputnik1969, 2018-06-13:
Hi,
In the above function there is this code:

    # when zero division occurs, use the limit value of the analytical expression
    cos_azi = np.where(np.isclose(denom,    0.0, rtol=0.0, atol=1e-8),  1.0, cos_azi)

    # when too many round-ups in floating point math take cos_azi beyond 1.0, use 1.0
    cos_azi = np.where(np.isclose(cos_azi,  1.0, rtol=0.0, atol=1e-8),  1.0, cos_azi)
    cos_azi = np.where(np.isclose(cos_azi, -1.0, rtol=0.0, atol=1e-8), -1.0, cos_azi)

my value of cos_azi is 1.0001652922628903 before going through these methods and is the same at the end i.e. cos_azi is not being set to a maximum of 1.0 and a minimum of -1.0.  This cause the arccos method to overflow.

Thanks
Ian

wholmgren, 2018-06-13:
What version of pvlib do you have installed? ``print(pvlib.__version__)``

adriesse, 2018-06-13:
You can't really mix and match sun path/solar position models, i.e. use zenith from one and azimuth from the other.  If the paths deviate from each other, which they do, then in general at a given point in time both zenith and azimuth will be different.  So the zenith angle you provided originally simply wasn't on the sun path defined by the the analytical model for that point in time.

I believe your experience confirms we did the right thing by using a small value for atol in the above code.

Sputnik1969, 2018-06-13:
OK - that confirms what I seen. Thanks.

closed 2018-06-13 by Sputnik1969