pvlib/pvlib-python #479: pvlib.pvsystem.calcparams_desoto

vanpnguyen, 2018-06-15:
Could someone help explain what cause the error and how to fix it?

I want to output Voc at a defined effective_irradiance and temp_cell.

Thank you,

[image]

cwhanse, 2018-06-15:
`pvlib.__version__` ?

cwhanse, 2018-06-15:
Ah, I think I see your problem.  You are calling `pvlib.pvsystem.calcparams_desoto` with the API from #471, but your pvlib package will have the old API for that function.  You can either go back to the old function call style (using the `module_parameters` dict or figure out how to copy the new `pvsystem.py` into your package. Sorry I can't be of more help with that last option.

wholmgren, 2018-06-15:
https://pvlib-python.readthedocs.io/en/stable/ links to the docs for the latest release.

https://pvlib-python.readthedocs.io/en/latest/ links to the docs for the latest commit on pvlib/master. I believe this is the default, and perhaps we should change that if possible.

You can google around for hints on installing the latest version from github in a way that fits your workflow. ``pip install <link to gh repo>`` is probably the easiest.

closed 2018-06-15 by wholmgren

mikofski, 2018-06-17:
@vanpnguyen seems like in the future this issue might be better asked at [pvlib-pyhthon Google group](https://groups.google.com/forum/m/#!forum/pvlib-python) thanks!