pvlib/pvlib-python #484: pvwatts loss model implementation with user defined input values

Jawairia123, 2018-06-18:
Hi, 

I am trying to use pvwatts loss model with ModelChain with user defined loss values instead of the default ones. 

I understand that for default values I just have to pass "pvwatts" string as the losses_model but I am not sure how can I link the loss model with user-specified values in the ModelChain function. I have tried all the possible ways that I can think of but still not sure how this will be implemented. 

Will appreciate if anyone can guide me to it!  

Thanks in advance! 

Regards

cwhanse, 2018-06-18:
You can use keyword arguments with `pvsystem.pvwatts_losses` .  It's not clear that these arguments are accessible through a `ModelChain` object - @wholmgren  can you confirm or correct?  If they are not accessible we need a new Issue opened.

wholmgren, 2018-06-18:
There is currently no easy way to combine ``ModelChain`` and user-defined losses for the pvwatts model. Implementing this in the standard pvlib pattern would require adding an attribute to the ``PVSystem`` object that defines the losses, then looking up that data during a call to ``PVSystem.pvwatts_losses``. For example, 

```
system = PVSystem(module_parameters={'pvwatts_losses' = {'soiling': 3, 'wiring': 1}}
system.pvwatts_losses()  # uses tools._build_kwargs to extract keys, if available.
```

One problem is that most of the losses parameters are poorly described by the existing ``PVSystem`` attributes (module_parameters, inverter_parameters, etc), so we might want to define a new generic attribute such as ``losses_parameters``. 

In the meantime, you can use the approach outlined [here](http://pvlib-python.readthedocs.io/en/latest/modelchain.html#User-defined-models) to define your own function that calls the existing ``pvsystem.pvwatts_losses`` with your parameters.

cwhanse, 2018-06-18:
> One problem is that most of the losses parameters are poorly described by the existing PVSystem attributes (module_parameters, inverter_parameters, etc), so we might want to define a new generic attribute such as losses_parameters.

I'm in favor of this approach.  Thinking further, perhaps `losses_parameters` could be a dict with entries for the different categories of loss.  We could start with the PVWatts list, or the PVsyst waterfall.  Either way, it would be an enhancement and merits its own Issue.

wholmgren, 2018-06-18:
I agree. It should be flexible enough to support other models, just like
module_parameters.

Jawairia123, 2018-06-19:
@wholmgren I tried linking my function with the `ModelChain`. 

If I include `"pvwatts"` as output to my function. `ModelChain` is accepting it but otherwise on linking when the output is an integer value, there appears an error (the first argument must be callable)

Jawairia123, 2018-06-19:
I have been suggested the following by one of the members of pvlib google group. 

`a = mc.ac * (1 - pvlib.pvsystem.pvwatts_losses(soiling=2, shading=3, snow=0, mismatch=2, wiring=2, connections=0.5, lid=1.5, nameplate_rating=1, age=0, availability=3)/100)`

i.e. to deduce the loss directly from the output. And since pvwatts applies losses only at final stage, this is working. 

It would be very helpful to see PVsyst system model and the loss calculation approach in pvlib-python.

PR #491 (2018-08-01): add PVSystem.losses_parameters, add support for custom pvwatts losses parameters in ModelChain
 - [x] Closes #484
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests must pass on the TravisCI and Appveyor testing services.
 - [x] Code quality and style is sufficient. Passes ``git diff upstream/master -u -- "*.py" | flake8 --diff`` and/or landscape.io linting service.
 - [x] New code is fully documented. Includes sphinx/numpydoc compliant docstrings and comments in the code where necessary.
 - [x] Updates entries to `docs/sphinx/source/ap [... cut]
closed 2018-08-01 by wholmgren