pvlib/pvlib-python #485: Invalid value encountered in power in clearsky

Spanner1969, 2018-06-20:
Hi,

I am using the get_clearsky method in the following way:

location = Location(50.430104, -5.006048, 'UTC', 0, 'newquay')
location.get_clearsky(times, model='ineichen', linke_turbidity=3)

with the time 1514764800000000000

I get the error: invalid value encountered in power (atmosphere.py:215)

Any ideas what is wrong?

Thanks
Ian

cwhanse, 2018-06-20:
Ian,

That code snippet works for me and produces
``
`Out[41]: `
`            ghi  dni  dhi`

2018-01-01  0.0  0.0  0.0`

wholmgren, 2018-06-20:
``Location.get_clearsky`` is designed to be called with a pandas DatetimeIndex. No promises if you call it with a different input type.

Spanner1969, 2018-06-20:
Yep - fair point. I did that and I get the same error:

location.get_clearsky(pd.DatetimeIndex([time, ]), model='ineichen', linke_turbidity=3)

I still get results - it is just an error thrown

Spanner1969, 2018-06-20:
It is a runtime warning - not an error as such but it does clearly point to an issue in the calculation

wholmgren, 2018-06-20:
A runtime warning does not necessary mean there is an error in the calculation. 

I do not get a runtime warning on:

Python 3.6.4
pandas 0.23
numpy 1.14.3
pvlib 0.5.2+6.ga3e05f2 (latest pvlib/master) but 0.5.* should be the same

What versions are you using?

cwhanse, 2018-06-20:
...and can you post the warning you see?  I didn't get any runtime warning.

Spanner1969, 2018-06-20:
This is script:

import pvlib
import pandas as pd
from pvlib.location import Location

location = Location(50.430104, -5.006048, 'UTC', 0, 'newquay')
location.get_clearsky(pd.DatetimeIndex([1514764800000000000, ]), model='ineichen', linke_turbidity=3)

I am using Python 2.7, pvlib 0.5 and numpy 1.14.3.

[image]

Spanner1969, 2018-06-20:
PVlib is actually version 0.5.2

cwhanse, 2018-06-20:
You are getting the warning because the sun is below the horizon at that time (zenith is -152 degrees). When that value is passed into the Kasten and Young air mass model, it results raising a negative number to the power of -1.643, hence the warning.

We could be more courteous about trapping these circumstances, and welcome a contribution here.

Spanner1969, 2018-06-20:
I guess that I could use a more recent version of Python. Would that solve the issue?

Spanner1969, 2018-06-20:
The reason that I say that is because of this code:

    # need to filter first because python 2.7 does not support raising a
    # negative number to a negative power.
    z = np.where(zenith > 90, np.nan, zenith)
    zenith_rad = np.radians(z)

cwhanse, 2018-06-20:
Perhaps.  I tried it with Python 3.6, not 2.7.  I wouldn't switch from 2.7 to 3.6 just for this warning.

mikofski, 2018-06-20:
But IMO you _should_ switch to Python-3.6 for many other reasons, mainly because support for Python-2 ends in 2020, and many packages are already not supporting Python-2 anymore

wholmgren, 2018-06-20:
I don't get the warning on 2.7.  It would help if you could reproduce the warning using ``atmosphere.relativeairmass`` directly since that's where the warning is coming from.

Spanner1969, 2018-06-21:
Python 3.6 does not seem to cause the same error so I am happy to upgrade. I will close this ticket.

closed 2018-06-21 by Spanner1969