pvlib/pvlib-python #488: remove irradiance parameter from ModelChain.run_model

wholmgren, 2018-06-23:
#239 deprecated the irradiance parameter in ``ModelChain.run_model``. It's time to remove it.

alamathe1, 2018-06-23:
@wholmgren Can I please help on this?

wholmgren, 2018-06-23:
sure, feel free to make a pull request. I should add that the parameter needs to be removed from ``ModelChain.prepare_inputs`` as well.

alamathe1, 2018-06-23:
@wholmgren Sure. I have removed the _irradiance parameter_ from ModelChain.run_model and ModelChain.prepare_inputs. Is there any preferred tests to be run for change or shall I go ahead and raise a PR?

alamathe1, 2018-06-23:
@wholmgren I have removed the irradiance parameter from ModelChain.run_model and ModelChain.prepare_inputs. Is there any preferred tests that needs to be run or shall I raise a PR?

wholmgren, 2018-06-23:
It looks to me like the ``pvlib/test/test_modelchain.py::test_weather_irradiance_input`` function needs to be removed, but otherwise no changes needed to the tests.

alamathe1, 2018-06-23:
@wholmgren Raised https://github.com/pvlib/pvlib-python/pull/490 Please let me know if there's any modifications required to the PR.

PR #490 (2018-06-26): remove irradiance parameter from ModelChain.run_model and ModelChain.prepare_inputs
pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python!

You may submit a pull request with your code at any stage of completion, however, before the code can be merged the following items must be addressed:

 - [x] Closes #488
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests must pass on the TravisCI and Appveyor testing services.
 - [x] Code quality and style is sufficient. Passes `` [... cut]
closed 2018-06-26 by wholmgren