pvlib/pvlib-python #493: NREL TMY server rejecting python requests

wholmgren, 2018-06-24:
**Describe the bug**
NREL's TMY file server recently started rejecting requests originating from urlopen. The response was:

```
"<html><head><title>Request Rejected</title></head><body>An error has occurred. Please consult with your administrator.<br><br>Your support ID is: 12437308730316847769<br><br><a href='javascript:history.back();'>[Go Back]</a></body></html>"
```

However, it's still possible to download a file from the same url with a web browser. Mostly likely they, intentionally or not, started blocking any requests that don't come with a browser-like HTTP User-Agent header.

I have a fix mostly ready to go.

**To Reproduce**
```python
pvlib/test/test_tmy.py F
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>> traceback >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

    @network
    def test_readtmy3_remote():
        url = 'http://rredc.nrel.gov/solar/old_data/nsrdb/1991-2005/data/tmy3/703165TYA.CSV'
>       tmy.readtmy3(url)

pvlib/test/test_tmy.py:19:

        # read in file metadata
        meta = dict(zip(head, csvdata.readline().rstrip('\n').split(",")))

        # convert metadata strings to numeric types
>       meta['altitude'] = float(meta['altitude'])
E       KeyError: 'altitude'

pvlib/tmy.py:178: KeyError
```

**Expected behavior**
Should work like it used to!

**Versions:**
all

closed 2018-06-26 by cwhanse

mikofski, 2018-07-26:
@wholmgren and @cwhanse unfortunately it looks like NREL has permanently retired the old rredc websites, however I think you can download a single CSV from MTS2:

```python
api_key = 'DEMO_KEY'
blah blah blah
```

what a PITA

cwhanse, 2018-07-26:
Agree with the comment about the NREL site limitations. The PSM NSRDB files have the same restrictions.

At what point does maintenance of this connection to the NSRDB become too much work?

adriesse, 2018-07-26:
Maybe if someone asked, they would keep it alive?

wholmgren, 2018-07-26:
> At what point does maintenance of this connection to the NSRDB become too much work?

I think the answer is "now". I suggest we remove the feature.

cwhanse, 2018-07-26:
> I think the answer is "now". I suggest we remove the feature.

I agree - we should remove the capability to pass a url to read_tmy3.

It was nice capability while it worked.

wholmgren, 2018-07-27:
NREL's TMY server is back online and the downloads are working again.

PR #494 (2018-06-26): specify user-agent in tmy3 remote request
 - [x] Closes #493 
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests must pass on the TravisCI and Appveyor testing services.
 - [x] Code quality and style is sufficient. Passes ``git diff upstream/master -u -- "*.py" | flake8 --diff`` and/or landscape.io linting service.
 - [x] New code is fully documented. Includes sphinx/numpydoc compliant docstrings and comments in the code where necessary.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and  [... cut]
commit e6aa17c5 (2018-06-26): specify user-agent in tmy3 remote request (#494)

Closes #493