pvlib/pvlib-python #503: GitHub doesn't recognize the license

mikofski, 2018-07-12:
**Describe the bug**
There should be a section near the top of the main repo overview page that shows the license
[image]
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'pvlib/pvlib-python' GitHub repository main page
2. Click on 'license'
3. Scroll down to 'anywhere'
4. See it's not there

**Expected behavior**
Users should be able to see the license immediately without having to look through the LICENSE document

**Screenshots**
Check out the license pages for pandas:
[github pandas repo](https://github.com/pandas-dev/pandas)
[pandas LICENSE file](https://github.com/pandas-dev/pandas/blob/master/LICENSE)
[image]

**Versions:**
 - ``pvlib.__version__``: any
 - ``pandas.__version__``:  any
 - python: any

**Additional context**
this might not bother some or most people

wholmgren, 2018-07-12:
Any idea why not? It’s standard bsd 3. Maybe the fact that it’s copyrighted Sandia in 2013 and pvlib developers after that?

mikofski, 2018-07-12:
yes I was comparing it to other bsd-3 licenses like pandas and pvmismatch and there is that difference, and maybe the missing bullets or numbers. Pandas also has line 1: "BSD 3-Clause License". [SunPower/PVMismatch](https://github.com/SunPower/PVMismatch/blob/master/LICENSE) license has no first line, but has numbers instead of bullets. The newlines don't seem to matter.

GitHub says this:
>If your repository is using a license that is listed on the Choose a License website and it's not displaying clearly at the top of the repository page, it may contain multiple licenses or other complexity. To have your license detected, simplify your LICENSE file and note the complexity somewhere else, such as your repository's README file.

uvchik, 2018-07-13:
You could try to re-create the file using an official github template and adapt it afterwards.

https://help.github.com/articles/adding-a-license-to-a-repository/

wholmgren, 2018-07-14:
Removing the second copyright line allowed github to recognize the license on my fork's master. Github suggests putting extra information in README.md. pandas split the copyright between license and AUTHORS.md. Either would work for me, though I lean towards AUTHORS.md. I think it's worth the effort/complexity so that we can do our part to improve the state of software licensing.

wholmgren, 2018-07-27:
Some examples of authors files:

https://github.com/pandas-dev/pandas/blob/master/AUTHORS.md
https://github.com/Unidata/MetPy/blob/master/AUTHORS.txt
https://github.com/dask/dask/blob/master/AUTHORS.md

I prefer the pandas model because it sounds like a pain to explicitly keep track of authors, either manually or with a tool. I will make a pull request following this model.

mikofski, 2018-07-28:
* I like pandas fix too
* IMO putting extra licence info in `AUTHORS.md` sense a little weird but, I don't want to overthink it, as is pvlib isn't using an `AUTHORS.md` file yet so why not
* IMO the way pandas is using `AUTHORS.md` is quite different from Unidata and Dask, since thos are just lists of contributors. In my experience keeping a list of contributors up to date is usually up to each contributor to add themselves, and I guess maintainers/owners police new contributors when merging as needed? Regardless, we as contributors to what's new instead, so, that was more than 2 cents.

PR #513 (2018-08-01): make github recognize license, add authors.md, clarify shared copyright

 - [x] Closes  #503
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and st [... cut]
closed 2018-08-01 by wholmgren