pvlib/pvlib-python #506: Correct condition 5 in `clearsky.detect_clearsky`

cwhanse, 2018-07-18:
**Describe the bug**
Condition 5 in `clearsky.detect_clearsky` does not agree with Eq. 12 - 14 of the reference.

**Expected behavior**
`clearsky.detect_clearsky` currently calculates condition 5 as the difference in maximum absolute slopes, rather than the maximum of the absolute differences in slopes.

**Versions:**
 - ``pvlib.__version__``: 0.5

wholmgren, 2018-07-18:
If I understand correctly, the pvlib code is essentially:

```
X  = max(abs(meas_slope)) - max(abs(clear_slope))
c5 = X < slope_dev
```

but Reno and Hansen describe:

```
X  = max(abs(meas_slope - clear_slope))
   = max(abs(line_diff_slope))
c5 = X < slope_dev
```

It would be good to include an example/test where this changes the overall result.

cwhanse, 2018-07-18:
@wholmgren you understand correctly.

Do we need to test before/after when fixing a bug though?

adriesse, 2018-07-18:
Perhaps not in general, but in this case it sound like the published results may have been based on code with a bug, so technically you should perhaps run the corrected code, check the results and publish an erratum.

wholmgren, 2018-07-18:
@cwhanse my feeling is that the [detect_clearsky_data.csv](https://github.com/pvlib/pvlib-python/blob/master/pvlib/data/detect_clearsky_data.csv) test data should include values that produce a different test result when the code is changed.

cwhanse, 2018-07-19:
Of course, I agree that we should correct the test along with the code fix.  I can publish a comparison on pvpmc.sandia.gov, or post the results here.

PR #596 (2018-10-09): Correct condition 5 in detect_clearksy
pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python! You may delete all of these instructions except for the list below.

You may submit a pull request with your code at any stage of completion.

The following items must be addressed before the code can be merged. Please don't hesitate to ask for help if you're unsure of how to accomplish any of the items below:

 - [x] Closes #506 
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/con [... cut]
closed 2018-10-09 by wholmgren