pvlib/pvlib-python #515: ENH: rename singlediode_methods.py to singlediode.py and discuss pvsystem.singlediode refactor

mikofski, 2018-08-03:
**Rename module name `singlediode_methods` and refactor `pvsystem.singlediode**
* `singlediode_methods` is a long name
* the suffix "methods" is ambiguous and not descriptive
* the other modules have meaningful names that correspond loosely to [the modeling steps described at Sandia PVPMC](https://pvpmc.sandia.gov/modeling-steps/)
* there needs to be a discussion regarding possibly refactoring `pvsystem.singlediode`
    - having a module shadow a method from a _different_ module might be confusing or maybe it's not a problem?
    - has there already been discussion on breaking `singlediode` up, refactoring, or renaming it?

**rename module `singlediode_methods.py` to singlediode.py and consider refactoring `pvsystem.singlediode`**
1. rename the module from `singlediode_methods.py` to just `singlediode.py`
2. change all references in the code where appropriate
3. change all references in the comments and documentation where appropriate
4. start a discussion on the repercussions to the `singlediode()` methods in `pvsystem.py`

**alternatives**
* rename the module something else like `dc_energy.py`, `pvmodules.py`, or `pvcells.py`
* do nothing
* rename the module to `singlediode.py` and leave `pvsystem.singlediode()` alone
* rename the module to `singlediode.py` and mode `pvsystem.singlediode()` to `singlediode.singlediode()` which is a more established paradigm, right?

**Additional context**
* this is a follow on to #504
* comments in #497
* also related to #408, #409, and #410

cwhanse, 2018-08-03:
My mental model has the following use cases:
- calculate IV curve parameters for a specific single diode model at given irradiance and temperature conditions.  This is the job of `calcparams_XXXX`, there's no way to avoid multiple functions because the single diode models use different sets of equations.
- calculate an IV curve for a given set of IV curve parameters.  This is the job of `singlediode` which should accept a `kwarg` to select a calculation method, e.g., `lambertw`, `bishop88`, others perhaps @thunderfish24. Although the standard single diode equation (SDE) and SDE+recombination are different, I am hopeful we can handle both calculations with the same set of methods: `bishop88` can and is the subject of #504, extending other methods to accommodate the recombination term is the job of this issue.
- `ModelChain` should have a method that conveniently wraps the various DC models, including `sapm`, `Desoto`, `CEC`, `PVsyst`. We have a partial implementation in `ModelChain.dc_model`.

Regarding #517, I vote strongly against `singlediode_pvsyst` unless we can't find any other workable way to host the several single diode models.

mikofski, 2018-08-08:
FYI: I'm on it right now, and BTW: Cliff was right, they clashed in `pvsystem.py`, but it's no problem.