pvlib/pvlib-python #516: ENH: calcparams_pvsyst should output d2mutau and vbi

mikofski, 2018-08-03:
**Problem**
* users may incorrectly use coefficients from `calcparams_desoto` with `d2mutau` in `singlediode`
* is there a way to restrict or limit use of `d2mutau` only for coefficients derived from `calcparams_pvsyst`?
* users may not correctly account for the correct number of cells in series and the the number of junctions per cell when determining builtin voltage parameter
* the `cells_in_series` argument in `bishop88` is redundant and inconsistent with the other argument with series cells `nNsVth`.

**Solution**
As @cwhanse suggested, replace `voltage_builtin` with `NsVbi` which is an output from `calcparams_pvsyst`, and `d2mutau` should also be passed through `calcparams_pvsyst` in order to discourage users from using it with CEC or DeSoto models.

>We currently pass the product `nNsVth` into `singlediode` and related functions, rather than passing `n`, `Ns`, and `Vth` separately. With the recombination current, the only place `Ns` is needed separately is with the per-cell `Vbi` to get a module `Vbi`. I think a more consistent method would be to pass `NsVbi` as the argument, rather than `Vbi`. That keeps all the single diode equation parameters as 'per-module' quantities, rather than a mix of per-cell and per-module.

**Alternatives**
Please suggest alternatives. One idea that was proposed originally was I think to lock in specific usage with the model, ie: have a separate `singlediode_desoto` and `singlediode_pvsyst` similar to the `calcparams_*` methods. Not sure how far down this needs to go, would it also apply to `bishop88`?

**Additional context**
this is an follow on to #504

adriesse, 2018-08-03:
How about supplementing each set of coefficients with a "source" identifier?  Then downstream functions can decide whether the "source" is suitable for them.
On the other hand this is a toolbox, and if someone want to use a hammer on a screw...

cwhanse, 2018-08-03:
I'm thinking about writing a `Module` class, which would contain datasheet information, parameter sets for the various performance models (which makes a sharp separation between CEC and PVsyst parameter sets), perhaps even measured IV curves. If we write that Class, part of it would be populated by the data in the SAM files.