pvlib/pvlib-python #517: ENH: model pvsyst thin-film recombination for CdTe and a:Si in pvsystem.singlediode()

mikofski, 2018-08-03:
**Problem**
This request was initially made in #163 and is also related to #470. As of #504 `bishop88` can accept the thin-film recombination parameters: `d2mutau` and `vbi` to output IV curve and its gradients, but there is no propagation of this ability to the proxies that use it such as `pvsystem.singlediode()`

**Solution**
Here is a proposed list of methods that should now be able to use these new terms:
* `singlediode_methods.bishop88_i_from_v()`
* `singlediode_methods.bishop88_v_from_i()`
* `singlediode_methods.bishop88_mpp()`
* `singlediode_methods._lambertw_v_from_i()`
* `singlediode_methods._lambertw_i_from_v()`
* `singlediode_methods._lambertw()`
* `pvsystem.PVSystem.singlediode()`
* `pvsystem.PVSystem.i_from_v()`
* `pvsystem.singlediode()`
* `pvsystem.max_power_point()`
* `pvsystem.v_from_i()`
* `pvsystem.i_from_v()`

**Alternatives**
Please suggest alternative solutions. There has been discussion of refactoring `calcparams_pvsyst` or even creating model specific `singlediode_pvsyst` methods. Is that desired?

**Additional context**
this is a follow on to #504

mikofski, 2019-09-11:
@adriesse didn't #762 and #763 already address this? I think this can be closed right?

- [x] [`pvsystem.max_power_point`](../pull/763/files#diff-150c57863db149a600771603bbec0ac1R2475)
- [x] [`singlediode.bishop88_i_from_v`](../pull/763/files?file-filters%5B%5D=.py#diff-923ebb302f2ea0f09560171028945713R177)
- [x] [`singlediode.bishop88_v_from_i`](../pull/763/files?file-filters%5B%5D=.py#diff-923ebb302f2ea0f09560171028945713R251)
- [x] [`singlediode.bishop88_mpp`](../pull/763/files?file-filters%5B%5D=.py#diff-923ebb302f2ea0f09560171028945713R324)

I think the other functions in the list above have a Lambert-W which I don't think has a way to handle the thin-film recombination terms yet.

Can I close this issue?

cwhanse, 2019-09-11:
I think this issue should stay open, or, be closed and replaced by several smaller issues that cover the scope:

1. finish implementation in pvsystem functions (see below)
2. implementation for PVSystem methods
3. implementation for ModelChain methods

Implementation in `pvlib.singlediode` is complete by #763. `bishop88` functions accept the arguments. The `lambertw` functions don't use the arguments and shouldn't; the method can't solve the single diode equation with the recombination term.

For the first item above, `pvlib.pvsystem.max_power_point` accepts the arguments by #763. The following functions could (and should) accept the arguments: 

-  `pvsystem.i_from_v`
-  `pvsystem.v_from_i`
-  `pvsystem.single_diode`

These functions have a `method` kwarg which defaults to `lambertw`, but could be set to `newton` or `brentq` and thus `d2mutau` could be used.