pvlib/pvlib-python #526: Nonzero direct irradiance contribution to sky diffuse when sun is behind the array

jdnewmil, 2018-08-09:
**Describe the bug**
In the `irradiance.py` module the `isotropic`, `klucher`, `haydavies`, `reindl`, and `king` functions have an `Rb` term that contributes an irradiance adjustment to the front-side diffuse even when the sun is behind the array or below the horizon. Under such conditions the only contributions to in-plane sky diffuse should be isotropic sky diffuse and (where modeled) horizon sky diffuse, and should not include the circumsolar component (particularly when this component is computed as a negative value).

**To Reproduce**
Run the code from the `test_total_irrad` function in `test_irradiance.py`. The second `ephem_data` row happens to represent this geometric condition as input... the test does not currently verify the values thus obtained, but if they are examined the behavior can be observed. For example, using the "haydavies" model the sky diffuse result is 20.4696793893283, but it should be 28.439461728635 because the `Rb` term is introducing a negative circumsolar irradiance contribution.

**Expected behavior**
Sky diffuse irradiance that does not vary significantly based on the sun apparent position when that position is behind the array or below the horizon.

I think that this can be fixed by setting `Rb=0` whenever `cos_tt` is negative or `cos_sun_zenith` is negative.

**Versions:**
 - ``pvlib.__version__``: '0.5.2+20.g09126e9'
 - ``pandas.__version__``: '0.23.1'
 - python: '3.6.6 |Anaconda, Inc.| (default, Jun 28 2018, 11:27:44) [MSC v.1900 64 bit (AMD64)]'

cwhanse, 2018-08-09:
Confirmed for `haydavies` and `reindl`. For `klucher`, the term involving `cos_tt` is squared so it has an erroneous effect but increases diffuse irradiance rather than decreasing.  The other sky diffuse models (`isotropic` and `king`) don't use a `cos_tt` term and don't specifically account for circumsolar irradiance.

This is an implementation error. #432 is relevant. We could either

1. fix in each sky diffuse implementation with `np.where(cos_tt<0)` or similar, or
2. change the AOI calculations to return 0 (or nan) when the sun is behind the array plane.

I'm inclined to the first option because I prefer the mathematically clean definition of AOI as the angle between the sun vector and the front-facing normal to the array, and I anticipate using AOI calculations with bifacial modeling.  In PVLib for Matlab, we implemented the first option.

wholmgren, 2018-08-09:
@jdnewmil thank you for catching and reporting this bug!

I also prefer to fix each diffuse implementation. 

I suggest that we fix this after #427 is merged in the next few days. The merge conflicts will relatively difficult to resolve if we do them in parallel.

We should fix #432 at the same time. 

The tests for the individual models do verify the returned values. With only a quick look, I believe that the second value in the expected results array is subject to this condition.

closed 2018-08-19 by cwhanse

cweickhmann, 2021-12-01:
Maybe related is a behaviour in `spectrum.spectrl2` that I [came across this week](https://stackoverflow.com/questions/70172766/pvlib-bird1984-north-facing-element-shows-negative-irradiance/70174010#70174010). North-facing panels exhibit negative irradiance. Not sure if this is due to the AOI value range or if the [NREL reference implementation](https://www.nrel.gov/grid/solar-resource/spectral.html) (cf. Bird1984) simply does not hold for purely indirect lighting.
I'll file a separate bug report with details and link it here.

PR #535 (2018-08-19): limit cos(aoi) in diffuse sky functions
Fixes two issues related to projection of the solar angle:

1. Dividing by near-0 cos(solar_zenith) when the sun is very close to the horizon.
2. non-0 contributions of the beam irradiance to the ``reindl``, ``haydavies``, and ``klucher`` functions.

Implementation is up for debate. Seemed easier to make a pull request with a concrete proposal.

I considered calculating the ``cos_solar_zenith`` as:

```python
cos_solar_zenith = aoi_projection(0, 0, solar_zenith, 0, projection_minimum=0.01745)
```

but I am worried this could lead to issues with small flo [... cut]