pvlib/pvlib-python #537: Irradiance calculation produces figures below real figures

Spanner1969, 2018-08-20:
I have the following script:

```
import itertools
import pandas as pd
import pvlib
from pvlib import clearsky, atmosphere
from pvlib.location import Location

cs = pvlib.location.get_clearsky_test(50.430104, -5.006048, 'ineichen', 
1525953600000000000, 'test')
print("ghi = " + str(cs['ghi'][0]))
print("dni = " + str(cs['dni'][0]))
print("dhi = " + str(cs['dhi'][0]))

# Parameters: Latitude, Longitude, Tilt, Azimuth, ghi, dni, dhi, time (Epoch)
result = pvlib.irradiance.globalinplane_test_output(50.430104, -5.006048, 
24, 180, cs['ghi'][0], cs['dni'][0], cs['dhi'][0], 1525953600000000000, .25, 
'grass')
print (result)
```
I have some wrapper methods here but the underlying methods will be called. This produces the following results:

```
ghi = 861.5440345805347
dni = 893.789882747542
dhi = 111.15085948276032
Solar Position azimuth =  172.79194271695087
Solar Position apparent_zenith =  32.90596711532058
AOI =  9.530542460995187
Airmass =  1.19028427148373
Gr-grass =  7.448439523040739
Time =  2018-05-10 12:00:00
Diff Perez =  125.92483957453679
Global in plane =                        poa_global  poa_direct  poa_diffuse
2018-05-10 12:00:00  1014.826609   881.45333   133.373279
poa_global  poa_direct  poa_diffuse
2018-05-10 12:00:00  1014.826609   881.45333   133.373279
```

The irradiance figure for the 10th May is about 10% lower than that measured by a Pyranometer at that location. From what I have read this model is generally lower than measured data.

Is there anything that I can try to improve the "theoretical" figures being generated as these seem lower than the real figures?

cwhanse, 2018-08-20:
@Spanner1969 thanks for the question. There are some alternate model choices which might close the gap between the model result and your measurements, but keep in mind, that irradiance models generally are 'average' models, over conditions and locations, and are not precise for any location or specific time. We default to the Ineichen clearsky model both for its simplicity and its performance relative to other simple models (see the Reno et al reference for the comparison).  Including altitude may help a bit, although the site you are modeling is already near sea level so this effect will be minor.

At some point in your modeling process you are transposing from horizontal to plane-of-array (probably with ` get_total_irradiance`) and that transposition defaults to the `isotropic` model, which is known to be biased towards low values.  You can select among other models for the transposition: I'd recommend `haydavies` for transposing modeled irradiance (and `perez` for transposing measurements).

Over a long period of time, a bias in modeled irradiance as compared to measurement could result from several reasons, including model bias and biased calibration of the irradiance instrument. Instrument issues might be identified and corrected by deploying redundant sensors. I've seen people address model bias by applying an empirical scalar to the clear sky irradiance. 

Please note that we have a google group pvlib-python for pvlib use questions, and we prefer to keep this forum for issues and discussion of the code itself.

@wholmgren we should add a link to the google group to the home page here.  We provide only a link to this Issues page currently.

closed 2018-08-20 by cwhanse

wholmgren, 2018-08-20:
@cwhanse are you suggesting that we replace the README.md's Contact section with the documentation's [Getting support](https://pvlib-python.readthedocs.io/en/latest/package_overview.html#getting-support) section? I am in favor of that.

cwhanse, 2018-08-20:
@wholmgren yes