pvlib/pvlib-python #545: bug in get_relative_airmass(model='youngirvine1967')

wholmgren, 2018-08-25:
**Describe the bug**
pvlib python implements the Young and Irvine 1967 model as:
```python
    elif 'youngirvine1967' == model:
        am = ((1.0 / np.cos(zenith_rad)) *
              (1 - 0.0012*((1.0 / np.cos(zenith_rad)) ** 2) - 1))
```

The published formula is

```
am = sec(z) * (1 - 0.0012 * (sec(z)^2 - 1))
```

See http://adsbit.harvard.edu/cgi-bin/nph-iarticle_query?bibcode=1967AJ.....72..945Y&db_key=AST&page_ind=3&plate_select=NO&data_type=GIF&type=SCREEN_GIF&classic=YES

If you squint very hard you might notice that pvlib python's implementation is equivalent to

```
am = sec(z) * (1 - 0.0012 * sec(z)^2 - 1)
```

**Versions:**
 - ``pvlib.__version__``: 0.5.2. I probably broke it when cleaning up code in the 0.1-0.2 era.

**Additional context**
Discovered when implementing #158. Write good unit tests! I'll make a pr with a fix soon.

cwhanse, 2018-08-25:
Nice squinting.

wholmgren, 2018-08-27:
closed by #546

closed 2018-08-27 by wholmgren