pvlib/pvlib-python #547: release 0.6.0

wholmgren, 2018-08-27:
We are close to a 0.6.0 release. I'd like to finish it this week. Here are the remaining issues and PRs labeled 0.6.0:

https://github.com/pvlib/pvlib-python/milestone/19

I'm ok with pushing all of these issues/prs to a later release. Please comment if you feel that any of those issues should be resolved first.

I'm going to try to address #514 today but I'm not sure how far I'll get.

cwhanse, 2018-08-27:
I'd like to resolve #417, #434 and #463 in 0.6.0.  I don't see that #421 can be closed any time soon, and I haven't started on #261. I'm OK pushing the remaining 0.6.0 issues out.

thunderfish24, 2018-08-28:
Push #411 too. I'm just back from a 2-week vacation, I'm pretty busy for September, and there have been changes to the single-diode models that I think will call for changes.

wholmgren, 2018-09-12:
This is the last remaining 0.6.0 issue. I'm ready to update the whatsnew and cut the final release. I'll make a PR for the whatsnew clean up in a few minutes. The alpha releases served the purpose of ensuring the deprecation tests are correct, so no need for a release candidate.

wholmgren, 2018-09-17:
Released!

closed 2018-09-17 by wholmgren