pvlib/pvlib-python #550: Irradiance.disc overflow error

Spanner1969, 2018-08-29:
Hi,

Running the following method gives me an overflow error and I am not sure of the best way to fix it:

time = 1520186400000000000
dni = irradiance.disc(38.283, 89.816079, time)

Any ideas on the best way to prevent this?

Thanks
Ian

wholmgren, 2018-08-29:
I'm guessing your time is integer nanoseconds since 1970. That's not a supported input type. Convert it to one of

```
datetime_or_doy : int, float, array, pd.DatetimeIndex

    Day of year or array of days of year e.g. pd.DatetimeIndex.dayofyear, or pd.DatetimeIndex.
```

closed 2018-08-29 by wholmgren

cwhanse, 2018-08-29:
That overflow warning is usually caused when the clearness index `kt` is too large (e.g., `kt=20`), which can happen at very low sun elevations with either measured or modeled GHI.  A limit on `kt` is introduced in #400 so v0.6.0 should not produce these warnings.  In the meantime, I think it is safe to ignore them.  The `disc` function should still return a value of 0.0, I believe.

cwhanse, 2018-08-29:
@wholmgren may not be supported but it works:

get_extra_radiation(time, 1370., 'spencer')
Out[15]: 1379.300811997068

wholmgren, 2018-08-29:
On further review... Cliff's explanation is likely correct for the error you're seeing. But I don't think ``disc`` can calculate ``kt`` correctly for that time input, so double check your code if this is not a one-off example.

wholmgren, 2018-08-29:
Sure, it does not error but I doubt it's the desired answer

```python
In [12]: time = 1520186400000000000

In [13]: time_ts = pd.Timestamp(time)

In [14]: time_ts
Out[14]: Timestamp('2018-03-04 18:00:00')

In [15]: time_ts.value
Out[15]: 1520186400000000000

In [16]: pvlib.irradiance.get_extra_radiation(time)
Out[16]: 1375.3743352329886

In [17]: pvlib.irradiance.get_extra_radiation(time_ts)
Out[17]: 1389.9028022507546
```