pvlib/pvlib-python #551: Add more irradiance metrics to pvlib.irradiance

kevinsa5, 2018-08-31:
The clearness index kt (the ratio of surface-level GHI to ET irradiance) can be calculated via `pvlib.irradiance.clearness_index` and is used in the DIRINT and DISC models.  Though other metrics in the same vein are not necessary for the irradiance models implemented in pvlib, including them in pvlib could be a convenience to users.  Is there interest in implementing other irradiance metrics for users to access?  Some examples:

- clear-sky index:  the ratio of surface GHI to modeled clear-sky GHI
- Variability Scores as described in this [Sandia paper](https://energy.sandia.gov/wp-content/uploads/dlm_uploads/2016/06/SAND2014-1638J-high_freq_solar_for_dist_studies_preprint.pdf) 
- Variability Index as described in this [Sandia paper, coauthored by Cliff Hansen](https://www.osti.gov/servlets/purl/1068417)

cwhanse, 2018-08-31:
I think `clearsky_index` is needed.  Would we pass the clear-sky GHI as an argument, or the name of a clear-sky model?  We'd have to agree how to deal with low irradiance conditions, e.g., an upper bound on `clearsky_index`. 

I think the variability calculations are interesting and would be worth adding as functions to the library.

wholmgren, 2018-08-31:
I am all for including these functions. Usually the best approach is to implement very simple functions that implement the core algorithm, then add convenience functions/methods around them. So, for clearsky index, I'd want a function that doesn't do any more than the division and limits (much like ``clearness_index``). That API would be straightforward and can be integrated into many modeling workflows. Consider something like ``Location.clearsky_index`` if you want the convenience of computing clear sky data within the function/method call.

Please do submit PRs!

cwhanse, 2019-03-06:
Closed via #571.  If we want to add the variability index, let's open a new issue.

closed 2019-03-06 by cwhanse