pvlib/pvlib-python #552: ENH: Implement PVSyst Cell Temperature Model

mikofski, 2018-08-31:
**PVSyst Cell Temperature Model**
For users using PVSyst model chain, there is no equivalent cell temperature model. The only cell temperature model is the Sandia array performance model, [`pvlib.pvsystem.sapm_celltemp`](https://pvlib-python.readthedocs.io/en/latest/generated/pvlib.pvsystem.sapm_celltemp.html). The [PVSyst cell temperature model](https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/cell-temperature/pvsyst-cell-temperature-model/) is different.

**Describe the solution you'd like**
Add a corresponding method like `pvlib.pvsystem.pvsyst_celltemp`. I'll bet it's already implemented in PVLIB_MATLAB, right?

**Describe alternatives you've considered**
leave it to the user to implement

```python
def pvsyst_celltemp(ambient_temp, alpha_absorption, gti, panel_efficiency, natural_convenction_coeff, forced_convection_coeff, windspeed):
    combined_convection_coeff = natural_convenction_coeff + forced_convection_coeff * windspeed
    return ambient_temp + alpha_absorption * gti * (1 - panel_efficiency) / combined_convection_coeff 
```

I'm not married to these names, they are only for demonstration

**Additional context**
* is this related to #487 at all? I tried to look for any corresponding or existing issues
* if this cell temperature model is implmented then should others be as well? what does SAM use?

cwhanse, 2018-08-31:
Great proposal @mikofski.  The [PVSyst temperature model](http://files.pvsyst.com/help/index.html) is   an extension of the [Faiman ](https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/module-temperature/faiman-module-temperature-model/) temperature model. We don't have either in PVLib MATLAB. They are easy enough to code but I don't know of a database of coefficients. Like the Sandia model, coefficient values depend both on the module construction and its mounting. PVSyst provides some defaults.

I read the recent bulletin board exchange between Mike Anderson and Andre Mermoud, it appears that there is still uncertainty regarding some of the PVsyst model terms (efficiency).

The Faiman model is prescribed by at least one IEC standard so it is of broad interest.  I would like to see this model added as well.  Different pull request, though.

How do people feel about changing the naming convention: `celltemp_sapm`, `celltemp_pvsyst` etc. rather than `sapm_celltemp` etc.?

wholmgren, 2018-08-31:
At the risk of issue scope creep... what about a ``celltemp.py`` module with functions ``sapm``, ``pvsyst``, etc.?

cwhanse, 2018-08-31:
Works for me

wholmgren, 2019-01-22:
Closed by #628.

closed 2019-01-22 by wholmgren

adriesse, 2019-01-22:
I have a PR coming...