pvlib/pvlib-python #553: clearsky.py requires PyTables but setup.py doesn't list it as install_requires

tadatoshi, 2018-09-04:
**Getting clearsky from Location causes an error**
I tried the [rdtools' excample](https://github.com/NREL/rdtools/blob/master/docs/degradation_example.ipynb), which used pvlib. 
At the first line under section "Clear Sky 0: Preliminary Calculations", executing 
clearsky = loc.get_clearsky(df.index, solar_position = sun)
causes an error:
ImportError: The Linke turbidity lookup table requires tables. You can still use clearsky.ineichen if you supply your own turbidities. 
Looking at the code found that it requires PyTables package. 
Installing PyTables by pip "pip install tables" solved the problem. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Location. e.g.
   loc = pvlib.location.Location(latitude, longitude, tz = timezone)
2. Get clearsky from loc. e.g.
   clearsky = loc.get_clearsky(df.index, solar_position = sun)
3. Execute the code

**Expected behavior**
The problem is that PyTables package is required but this is not documented, nor written in setup.py. 
This should be documented and PyTables should be lisetd as install_requires in setup.py.

**Versions:**
 - ``pvlib.__version__``: 0.5.2
 - ``pandas.__version__``: 0.23.4
 - python: 3.6.4

**Additional context**
This is not necessarily a bug. But other categories didn't seem to fit thus I put this as a Bug report.

wholmgren, 2018-09-04:
Thanks for posting. The base pvlib installation does not require packages such as pytables and scipy that are only used in a handful of functions. See [Installation/Compatibility](https://pvlib-python.readthedocs.io/en/latest/installation.html#compatibility).

See #483 for related discussion about making scipy a requirement.

cwhanse, 2018-09-04:
I think something else is amiss in your code @tadatoshi 

This runs without error for me:

`
import pvlib
import pandas as pd

dt = pd.DatetimeIndex(start='2018-08-30', end='2018-08-31', freq='1H')

df = pd.DataFrame(index=dt)

latitude = -23.762
longitude = 133.874886
timezone = 'Australia/North'
df.index.tz_localize(timezone)

loc = pvlib.location.Location(latitude, longitude, tz = timezone)

sun = loc.get_solarposition(df.index)

clearsky = loc.get_clearsky(df.index, solar_position = sun)
`

wholmgren, 2018-09-04:
@cwhanse I believe you are using the base Anaconda environment which already includes pytables. It sounds like @tadatoshi uses pvlib in an environment that is not prepopulated with pytables, scipy, etc. If I understand correctly, @tadatoshi expects that ``pip install pvlib`` will automatically install pytables if it's not already installed.

cwhanse, 2018-09-04:
@wholmgren you are correct

tadatoshi, 2018-09-05:
Thank you for prompt responses. 
I appreciate it very much. 

I write the context that led me to use pip instead of Anaconda. 
I was trying to use [RdTools](https://github.com/NREL/rdtools) in Jupyter Notebook started in Anaconda environment. 
However, "conda install rdtools" failed because it was not found in the default Anaconda repositories. 
I couldn't find their documentation about how to install it in Anaconda. 
Hence, I decided to run Jupyter Notebook from Virtual Environment based on pip. 
I installed all the packages specified in setup.py of pvlib-python and jupyter, matplotlib and rdtools in the virtual environment. 
Then encountered the error because I didn't install PyTables because I didn't know that it was required. 

Maybe this is a special case. 

Please feel free to close this issue since the new comment on issue #86 addresses this issue. (The link to this issue is also added there.)

PR #566 (2018-09-11): add extras_require complete and test options
 - [x] Closes #553
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style is sufficient. Passes LGT [... cut]
closed 2018-09-11 by wholmgren