pvlib/pvlib-python #567: readthedocs are failing

mikofski, 2018-09-07:
**Describe the bug**
See https://readthedocs.org/projects/pvlib-python/builds/

**To Reproduce**

<details>

```
Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/nbsphinx.py", line 653, in parse
    rststring, resources = exporter.from_notebook_node(nb, resources)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/nbsphinx.py", line 575, in from_notebook_node
    nb, resources = pp.preprocess(nb, resources)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/nbconvert/preprocessors/execute.py", line 359, in preprocess
    with self.setup_preprocessor(nb, resources, km=km):
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/contextlib.py", line 81, in __enter__
    return next(self.gen)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/nbconvert/preprocessors/execute.py", line 304, in setup_preprocessor
    self.km, self.kc = self.start_new_kernel(cwd=path)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/nbconvert/preprocessors/execute.py", line 253, in start_new_kernel
    km.start_kernel(extra_arguments=self.extra_arguments, **kwargs)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/jupyter_client/manager.py", line 246, in start_kernel
    kernel_cmd = self.format_kernel_cmd(extra_arguments=extra_arguments)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/jupyter_client/manager.py", line 170, in format_kernel_cmd
    cmd = self.kernel_spec.argv + extra_arguments
AttributeError: 'NoneType' object has no attribute 'argv'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/sphinx/cmdline.py", line 304, in main
    app.build(args.force_all, filenames)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/sphinx/application.py", line 331, in build
    self.builder.build_update()
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/sphinx/builders/__init__.py", line 342, in build_update
    'out of date' % len(to_build))
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/sphinx/builders/__init__.py", line 355, in build
    updated_docnames = set(self.env.update(self.config, self.srcdir, self.doctreedir))
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/sphinx/environment/__init__.py", line 568, in update
    self._read_serial(docnames, self.app)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/sphinx/environment/__init__.py", line 587, in _read_serial
    self.read_doc(docname, app)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/sphinx/environment/__init__.py", line 662, in read_doc
    doctree = read_doc(self.app, self, self.doc2path(docname))
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/sphinx/io.py", line 302, in read_doc
    pub.publish()
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/docutils/core.py", line 217, in publish
    self.settings)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/docutils/readers/__init__.py", line 72, in read
    self.parse()
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/docutils/readers/__init__.py", line 78, in parse
    self.parser.parse(self.input, document)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pvlib-python/conda/latest/lib/python3.6/site-packages/nbsphinx.py", line 664, in parse
    str(e))
nbsphinx.NotebookError: AttributeError in modelchain.ipynb:
'NoneType' object has no attribute 'argv'

Notebook error:
AttributeError in modelchain.ipynb:
'NoneType' object has no attribute 'argv'
```

</details><br/>

see https://readthedocs.org/projects/pvlib-python/builds/7752730/

I wonder if the `python` in `docs/environment.yml` should be Python-3?

Or maybe just use virtualenv with pip since the rtdf server is linux?

**Expected behavior**
there should be some v0.6.0-alpha docs or latest.

**Screenshots**
- https://readthedocs.org/projects/pvlib-python/builds/7752730/
- https://readthedocs.org/projects/pvlib-python/builds/

**Versions:**
 - v0.6.0-alpha
 - python: 2 or 3 hard to tell

**Additional context**
there are also some other random errors due to conda which has experimental support, why not just use a virtualenv with pip, should be a lot more reliable, IMO

wholmgren, 2018-09-07:
> I wonder if the python in docs/environment.yml should be Python-3?

There used to be a reason for keeping it on python 2, though I forget what it was. Probably doesn't matter with new versions of packages/rtd.

> why not just use a virtualenv with pip, should be a lot more reliable, IMO

This was not possible when we had to compile the packages on rtd. Might work now.

No reason to blame readthedocs for this failure, though. It's due to some new incompatibility among the required packages.

wholmgren, 2018-09-07:
something to do with ``nbsphinx``. It works (mostly) when I manually convert the ``modelchain.ipynb`` file to ``rst``. I suggest we resolve this by replacing the ipynb file with an rst file and ending our experiment with nbsphinx.

PR #568 (2018-09-09): convert modelchain.ipynb to rst, remove nbsphinx
 - [x] Closes #567 
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [ ] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style is sufficient. Passe [... cut]
closed 2018-09-09 by wholmgren