pvlib/pvlib-python #569: incorrect horizon test in tracker.singleaxis

cwhanse, 2018-09-10:
**Describe the bug**
`    wid[zp <= 0] = np.nan` sets the tracker rotation angle to `np.nan` when the sun is behind the array plane, but the array plane is not yet defined. At this line, the array plane is defined only by the tracker tilt and azimuth, but not its rotation.

**To Reproduce**
```
import pvlib
import numpy as np
import pandas as pd
import matplotlib.pyplot as plt

loc = pvlib.location.Location(latitude=30, longitude=110)
sat = pvlib.tracking.SingleAxisTracker(axis_tilt=90, axis_azimuth=180, backtrack=False, max_angle=180)
sat.localize(location=loc)

solar_azimuth = pd.Series(data=np.arange(0.0, 359.0))
solar_zenith = 45.0 * solar_azimuth / solar_azimuth

df = sat.singleaxis(apparent_zenith=solar_zenith, apparent_azimuth=solar_azimuth)

p1 = plt.plot(df)
plt.legend(df.columns)
```
The plot shows that the tracker rotation `tracker_theta` is limited to between -90 and 90.

**Expected behavior**
In the above example, the sun rotates in azimuth at 45 zenith. The vertical tracker without rotation limit should follow the sun around the complete circle.

**Screenshots**
[image]

**Versions:**
 - ``pvlib.__version__``: 0.5.2 

**Additional context**

One fix would be to substitute a test for solar zenith, so that the tracker rotation is not set when the sun is below the visible horizon. I also wonder if we should set the rotation to 0 (stow position)  instead of `np.nan` in this case.

`    wid[z <= 0] = np.nan`
`    wid[z <= 0] = 0.0`

wholmgren, 2018-09-10:
Other than 0 vs. nan, I believe this follows ``pvl_singleaxis.m`` lines 167-171:

```matlab
% filter for sun above panel horizon)
u = zp>0;

% apply limits to ideal rotation angle
wid(~u) = 0;  % set horizontal if zenith<0, sun is below panel horizon
```

Or did I mix things up when implementing ``singleaxis``?

On 0 vs. nan, I believe my thinking at the time was that it might not be safe to always assume a 0 stow position and in any case a user could always replace the nan with 0s if they cared. I'm generally ok with changing it, though.

wholmgren, 2018-09-10:
Ok, so ``singleaxis`` lines 413-414

```python
    # filter for sun above panel horizon
    wid[zp <= 0] = np.nan
```

become

```python
    # filter for sun above horizon
    wid[solar_zenith >= 90] = np.nan  # or 0 if you prefer
```

Sounds good to me. 

Would be best to add a new test that uses the same input data as above but with the ``singleaxis`` function rather than the object method.

wholmgren, 2018-09-11:
@cwhanse I can do this this afternoon if you're not already on it. Would like to finish off 0.6.0...

cwhanse, 2018-09-11:
I'm not free this afternoon, so go ahead and fix this bug if you can

adriesse, 2018-09-12:
On the topic of "zero vs nan": I wonder whether we could maintain a master list somewhere listing all the cases where a value is special, such as being out of range or non-physical, and how it is encoded, such as zero, nan, -99.99, 7999, or whatever.

PR #573 (2018-09-12): fix incorrect nan assignments in singleaxis, allow scalar and 1d array input
 - [x] Closes #569
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and [... cut]
closed 2018-09-12 by wholmgren