pvlib/pvlib-python #577: Implement another snow loss model like Bill Marion as an alt to pvwatts

mikofski, 2018-09-13:
I'll try to track down the papers on this

anomam, 2019-04-16:
Hello folks, @mikofski @wholmgren @cwhanse 
My group would like to open-source in pvlib the snow model that Defne Gun presented at the 10th PVPMC workshop in 2018: see [link](https://pvpmc.sandia.gov/resources-and-events/events/2018-10th-pv-performance-modeling-workshop/) to workshop website and [link](https://pvpmc.sandia.gov/download/6446/) to presentation.
I also talked with Dan Riley (do you know his Github username? I would like to tag him) and it seems that it would be very useful for ongoing research to have that model implemented so that his group can build on top of it.
Would that be ok? Please let us know if there are any concerns.

mikofski, 2019-04-16:
ICYMI @patrickjmathiesen

mikofski, 2019-04-16:
I'd still like to see Bill Marion's model here,  but I think it would be great to see the SunPower model too,  thanks!

DanRiley, 2019-04-16:
I think that Defne's additions are just on top of Bill's model, so I'm hopeful that by getting Defne's model that we would also get Bill's. I'll be working on getting both of them into MATLAB pv_lib.

cwhanse, 2019-04-16:
Thanks for the initiative on this. Two questions to help identify where to put the code:
- Does the model modify irradiance or power?
- What system characteristics are needed beyond tilt and azimuth?  It appears that a length dimension and height above ground are needed.

cwhanse, 2019-06-25:
@ryanblumie @kevinsa5 Ryan let's discuss your offer here

mikofski, 2019-06-26:
FYI: DNV GL would also like to contribute the Townsend snow model, I have 3 references. The inputs are typically daily snowfall history, gcr, module lower edge height above ground, ambient temperature,  and the output is binary: no power direw to snow cover, or power. AFAIK there is no partial power. We already have the code ready.

cwhanse, 2019-06-26:
@mikofski inputs are similar to Marion's model. As far as pvlib, we need to decide where to record the inputs and how to name them. Tossing one idea out:
`PVSystem.height_above_ground`
`PVSystem.gcr` - need to move from `SingleAxisTracker.gcr`
Daily snowfall doesn't fit nicely in `weather` due to the expected daily timesteps. Ideas for a nice way to host snowfall/rainfall other daily values (turbidity perhaps)?

wholmgren, 2019-06-26:
> Daily snowfall doesn't fit nicely in weather due to the expected daily timesteps. Ideas for a nice way to host snowfall/rainfall other daily values (turbidity perhaps)?

Is this only a concern for integrating the snow loss model with `ModelChain.run_model(weather)` or do we need to address it when implementing the basic function?

cwhanse, 2019-06-26:
The concern is for `ModelChain`  The basic function should expect to get the inputs that the model requires.

PR #764 (2020-03-28): Snow coverage model from SAM
- [x] Closes #577 
- [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style is sufficient. Passes LGTM and SticklerCI  [... cut]
closed 2020-03-28 by CameronTStark