pvlib/pvlib-python #586: Incorrect reference in forecast.cloud_cover_to_ghi_linear

cwhanse, 2018-09-20:
**Describe the bug**
The comment text for `forecast.cloud_cover_to_ghi_linear` contains this reference

>         References
>         ----------
>         Larson et. al. "Day-ahead forecasting of solar power output from
>         photovoltaic plants in the American Southwest" Renewable Energy
>         91, 11-20 (2016).
> 

The linear relationship asserted between cloud cover and transmittance does not appear in this paper.  It may be that the linear model presented for GHI to power output (PO) is being cited by mistake.

wholmgren, 2018-09-20:
I'm confused. Does it not implement equation 2 (with the default offset of 35)?

```python
        offset = offset / 100.
        cloud_cover = cloud_cover / 100.
        ghi = (offset + (1 - offset) * (1 - cloud_cover)) * ghi_clear
```

cwhanse, 2018-09-20:
It does, the confusion is mine.

closed 2018-09-20 by cwhanse