pvlib/pvlib-python #587: Curve IV regression

tcapelle, 2018-09-21:
I wanted to ask if the pvlib library can fit IV curves, from IV measures to params?

I am currently working with data from our station here at INES and would like to be able to retrieve the curve params (Iph, I0, etc...) from data with shadows (and without).
I know the Matlab code has the feature, but it is not very stable.
Any insights on this topic? Maybe a Deep Learning approach? I would like to cooperate with the pvlib development of this feature.
Sincerely,
Thomas

cwhanse, 2018-09-21:
At present, no.  There is a stale pull request #229 to implement the algorithm from Matlab, and an open issue #511 asking for another method.  Both are intended to return performance model parameters (PVsyst and CEC, respectively) rather than fit a single IV curve, but fitting a curve is a necessary part of the process.

A contribution here would be very welcome.  The challenge is that there are literally hundreds of published algorithms for determining diode equation coefficients for an IV curve.  In my opinion, many are not validated, rather they are shown to produce output for a few test IV curves that looks reasonable.  A validated algorithm would recover known parameter values from IV curves calculated using the single diode equation, even in the presence of simulated measurement noise. In my view, a successful pull request requires 1) a published description of the algorithm, and 2) validation as described.

Separately, I'm very interested to hear what issues you have found with the Matlab algorithm.  It is probably best if those issues are raised on the [PVLib Matlab github](https://github.com/sandialabs/MATLAB_PV_LIB)

thunderfish24, 2018-09-21:
@tcapelle When you say data with shadows, do you mean that you want the algorithm to filter out or flag those data as not good? Many standard diode models make homogeneity assumptions that are violated by shading. Are you considering only the diode models currently in pvlib-python?

mikofski, 2018-09-22:
@cwhanse can you recommend some good references for parameter estimation? I have a few, and I've started to study the pvl_matlab code. From my experience so far, a lot seems to depend on my ability to estimate a good initial guess, has that been your experience?

thunderfish24, 2018-09-23:
@mikofski I offer no formal evidence, but I seem to have found much more robust convergence w.r.t. choice of initial conditions by using entire I-V curves rather than 5 or so “select” points. That said, a well-chosen IC always helps ;).

thunderfish24, 2018-09-24:
@tcapelle I recommend trying my own [pvfit.app](pvfit.app) for the single-diode model (SDM), with reference [here](https://onlinelibrary.wiley.com/doi/full/10.1002/ese3.190). It can fit a "local" SDM to curves at one irradiance and temperature (SDM-L) and a particular "global" SDM to sets of curves at different irradiance and temperature (SDM-G). Note that the double-diode model (DDM) at PVfit is still very much beta-ware, and I am working on fixing a problem with the ODR solver I've been using, which doesn't seem to work well on certain DDM-L fits.

I also recommend http://pvmodel.umh.es/, which seems to provide somewhat different (perhaps better?) fits to the SDM in the face of model discrepancy (aka, model error, i.e., SDM is the wrong model). For an example, of that discrepancy, see the figure I posted [here](https://github.com/SunPower/PVMismatch/issues/88).

cwhanse, 2018-09-24:
@mikofski for the Matlab algorithm I have a [Sandia technical report](https://prod.sandia.gov/techlib-noauth/access-control.cgi/2015/152065.pdf).

For a survey of methods to extract electrical parameters from IV curves: [Ortz-Conde](https://pdfs.semanticscholar.org/13c0/f8796999231d6cfcf275e8cd3bbaf69499de.pdf)

From my perspective, methods fall into two broad categories (there are overlaps):

1.  fit the IV curve equation to the data (using a few points, the whole curve, all different kinds of objective functions and numerical approaches). The parameter values thus obtained depend on each other and may not be constrained by the physics being modeled.
2. determine individual parameters from a reduced order model (e.g., Rsh from the IV curve slope near Isc, diode factor from the relationship between Voc and log(irradiance)).

The NREL approach for the CEC model coefficients is of type 1 (find the zero of a set of 6 equations, written in units of amps). PVfit is also type 1.  Aron told me that in SAM they use the diode factor estimator (Voc vs log(irradiance)) in SAM to reduce the order to 5, so that is a mix of type 2 and type 1.  The PVLib Matlab algorithm is also a mix of the two types.

cwhanse, 2018-09-24:
> I am interested of reconstructing the curve without shading. Is the Pvlib matlab repo active, it looks dead...

@tcapelle [Pvlib matlab](https://github.com/sandialabs/MATLAB_PV_LIB)  isn't dead but doesn't get much traffic, because most people find PVLib matlab on [PVPMC](pvpmc.sandia.gov) and email us with questions.

tcapelle, 2018-09-26:
Thanks @cwhanse , I am looking at the Matlab code, and the idea of using the CoContent function is pretty clever.
You get some regularisation just for using the integral, I may try this approach for my DL parameter estimation.
Transfering the Matlab `est_single_diode_param` function to Python looks very straight forward, I may give it a try.

cwhanse, 2018-09-26:
Regarding #229, if I recall correctly, the missing work is unit tests for the new functions.

cwhanse, 2020-09-06:
Closed by #708 although there is certainly room for other fitting methods, and additions are welcome.

closed 2020-09-06 by cwhanse