pvlib/pvlib-python #589: add UO SRML data reader

wholmgren, 2018-09-21:
Add data reader for the University of Oregon Solar Radiation Monitoring Laboratory archive http://solardat.uoregon.edu/

Primarily interested in the 1 minute data sets. Challenges include

- translating numeric columns to pvlib names
- stitching together files across months
- stitching together files when the data format at a station changes
- determining metadata for a station

This should be a new module under the `pvlib.iotools` subpackage proposed in #272. 

cc @lboeman

lboeman, 2018-09-26:
A couple more challenges I have discovered:
- Some files/stations have multiple values for the same variable recorded by different instruments. For instance, the file from Eugene below has 6 ghi measurements and 3 dni (100x and 201x respectively).  
- It seems both -999 and -99.9 exist as fill values. For example, ambient temperature (9303) has a fill value of -99.9 and dhi (3001) has a fill value of  -999 in this file: http://solardat.uoregon.edu/download/Archive/EUPO1704.txt

wholmgren, 2018-09-26:
We might want to map all `100[1-9]` values to `ghi_[1-9]`, all `201[1-9]` to `dni_[1-9]`, etc. It would be convenient to also get a simple set of names like `ghi` and `dni` but the feasibility of that  depends on the failure modes of the data. Maybe median of all values for which the QA flag is ok. So a file with 3 GHI columns labeled `1001, 1002, 1003` would produce a DataFrame with 4 columns: `ghi_1, ghi_2, ghi_3, ghi` where `ghi = data.filter(like='ghi').median()` 

@cwhanse @adriesse are there recommended algorithms for how to combine multiple measurements of the same irradiance quantities from a single location? 

Probably ok to set nans like `data[data < -99] = np.nan`.

cwhanse, 2018-09-26:
depends on the instruments in use. For the same quality/type of instrument, it makes sense to average the measurements (regarding the differences among instruments as random error). It does not make sense to average among different instruments (e.g., photodiodes and thermopile pyranometers).

wholmgren, 2018-09-26:
Good point on instrument type. I don't see a simple way to handle this for the UO data set based only on the metadata on their website. Maybe they have more available offline. For now I suggest we move forward with a PR that simply maps the tsv columns like `1001, 1002, 1003` to a DataFrame with columns like `ghi_1, ghi_2, ghi_3`.

PR #594 (2018-10-02): Add UO SRML data reader
pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python! You may delete all of these instructions except for the list below.

You may submit a pull request with your code at any stage of completion.

The following items must be addressed before the code can be merged. Please don't hesitate to ask for help if you're unsure of how to accomplish any of the items below:

 - [x] Closes #589 
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html) [... cut]
closed 2018-10-02 by wholmgren