pvlib/pvlib-python #598: solarposition.hour_angle is inefficient and fails on some pandas versions

wholmgren, 2018-10-04:
**Describe the bug**
For some reason related to timezone handling, the `solarposition.hour_angle` function fails on pandas versions 0.15-0.18. We test 0.14 and latest. Apparently we were on pandas 0.19 or greater when this function was created.

Also, the loop over datetime values is probably slow.

```python
    hours = np.array([(t - t.tz.localize(
        dt.datetime(t.year, t.month, t.day)
    )).total_seconds() / 3600. for t in times])
```

I think this gives the same result:

```python
hours = (times.astype(int) - times.normalize().astype(int)) / 1e9 / 3600
```

**Versions:**
 - ``pvlib.__version__``: 0.6.0
 - ``pandas.__version__``: 
 - python:

**Additional context**
probably should address #597 first

mikofski, 2018-10-04:
@wholmgren I believe this may be related to the pandas issues I observed in #583 - I noticed this `hour_angle` code, and I thought I should have addressed it then, but I didn't. It was my bad code from the Bird clear sky model, sorry!

wholmgren, 2018-10-04:
No worries, Mark. Do you want to address this and #597 in #583? Or do you think we should handle them separately?

mikofski, 2018-10-04:
I will address them, but I'd rather open a new PR to fix the `hour_angle` issues, unless you would prefer not

wholmgren, 2018-10-04:
New PR sounds good to me.

PR #599 (2018-10-05): add test for hour_angle, vectorize
* closes #598 vectorize to make it more efficient
* closes #597 add test

pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python! You may delete all of these instructions except for the list below.

You may submit a pull request with your code at any stage of completion.

The following items must be addressed before the code can be merged. Please don't hesitate to ask for help if you're unsure of how to accomplish any of the items below:

 - [ ] Closes issue #xxxx
 - [ ] I am familiar with th [... cut]
commit 5ee82b1c (2018-10-05): add test for hour_angle, vectorize (#599)

* add test for hour_angle, vectorize

* closes #598 vectorize to make it more efficient
* closes #597 add test

* BUG: use np.int64 works better for older numpy version than python int

* also converting times to int before subtracting works better for older
pandas versions which were not calculating the timedeltas correctly
* remove comment with missing space after hash, add FIXME that explains
why the expected values are slightly different than the SPA calculator
output

* fix hanging indent

* BUG: replace utcoffset() with reliable, efficient appro [... cut]
closed 2018-10-05 by wholmgren