pvlib/pvlib-python #601: add NREL MIDC reader to iotools

wholmgren, 2018-10-10:
It might be useful to add a `pvlib.iotools` parser for the NREL MIDC sites, particularly for the 1 minute data from sites that are currently active. 

I've copied/pasted code similar to this more times than I'd care to admit:

```python
url = 'http://midcdmz.nrel.gov/apps/plot.pl?site=UAT&start=20101103&edy=2&emo=4&eyr=2018&year=2016&month=1&day=1&endyear=2017&endmonth=12&endday=31&time=23:59&inst=4&inst=5&inst=6&inst=7&inst=9&type=data&first=3&math=0&second=-1&value=0.0&user=0&axis=1'
weather = pd.read_csv(url)
datetime = weather['DATE (MM/DD/YYYY)'] + weather['MST']
datetime = pd.to_datetime(datetime, format='%m/%d/%Y%H:%M')
weather = weather.set_index(datetime)
weather = weather.tz_localize('America/Phoenix')
weather = weather.rename(columns={'Global Horiz (platform) [W/m^2]': 'ghi', 
                                  'Direct Normal [W/m^2]': 'dni', 
                                  'Diffuse Horiz [W/m^2]': 'dhi',
                                  'Air Temperature [deg C]': 'temp_air',
                                  'Avg Wind Speed @ 3m [m/s]': 'wind_speed'})
weather = weather.drop(columns=['DATE (MM/DD/YYYY)', 'MST'])
```

Like the UO SRML network, the NREL network has different instruments at different sites, and sometimes the instruments change.

NREL has a fairly straightforward [API for daily data](https://midcdmz.nrel.gov/apps/data_api_doc.pl). I don't know if it's useful to try to wrap that in addition to the 1 minute data. It could help determine instrument changes.

There's also a nice site API here: https://midcdmz.nrel.gov/apps/data_api_doc.pl?_idtextlist_

**Describe alternatives you've considered**
Let users implement their own MIDC parsers as needed.

**Additional context**
Could be useful for Solar Forecasting 2. cc @cwhanse @lboeman @alorenzo175

cwhanse, 2018-10-17:
Perhaps #605 isn't the PR to do this, but, do we provide the pvlib variable names someplace?  We ask users to create a dict mapping MIDC to pvlib names, it would help to point users to the list of available pvlib names.

lboeman, 2018-10-17:
I believe they are provided  [pvlib/data/variable_style_rules.csv](https://github.com/pvlib/pvlib-python/blob/master/pvlib/data/variables_style_rules.csv). and rendered in the documentation at [https://pvlib-python.readthedocs.io/en/latest/variables_style_rules.html](https://pvlib-python.readthedocs.io/en/latest/variables_style_rules.html).
Although I found it by accident myself, so it's probably a good idea to point users in the right direction.