pvlib/pvlib-python #602: Ecmwf reader of AOD and Pwat reader for iotools

mikofski, 2018-10-11:
I have a reader from [pvsc44](../../../mikofski/pvsc44-clearsky-aod/blob/master/all_ecmwf_macc.py) that makes an ["atmosphere" object](https://github.com/mikofski/pvsc44-clearsky-aod/blob/master/pvsc44_clearsky_aod/ecmwf_macc_tools.py). Sh old something like this be added to `iotools`?

cwhanse, 2018-10-17:
I'm not opposed to adding functions that read from ECWMF files, but @wholmgren do we want a new object, or a `iotools.ecmwf` module?

wholmgren, 2018-11-02:
Sorry I missed this until now. That seems like a useful thing to add to `iotools`. I would quickly merge a PR with a scope limited to reading AOD data from MACC. `iotools.ecmwf` sounds like a bigger design/scope challenge. In a similar spirit, I'd prefer a more specific class name such as `ECMWF_AOD_PW`.

mikofski, 2018-11-04:
Okay, I thinking I can do that, and quite frankly I'd rather limit this to AOD and Pwat, because the amount of data at soda pro and ecmwf is staggering, and I wouldn't know where to start.

>In a similar spirit, I'd prefer a more specific class name such as `ECMWF_AOD_PW`

I wasn't sure what you meant. It's all caps, it's this a constant? In lieu of `Atmosphere`?

wholmgren, 2018-11-04:
Class name, not a constant. How do you name python class that’s an
acronym???

Atmosphere could mean a lot of things if it’s associated with weather model
data.

mikofski, 2018-11-04:
Q: should iotools assume that the user has already obtained the data from the ecmwf API and just parse the files? Navigating their API is a challenge, at least for me. Is it okay to include downloading the data in iotools too somehow? Like `download_ecmwf_aod_pwat()`?

wholmgren, 2018-11-04:
Up to you - I’m trying to take a more flexible approach to iotools. You can
check out our recent iotools additions for some examples of how we tried to
decouple the core parsing features from the api navigation features.

mikofski, 2018-11-11:
FYI: I was queued for an hour on their server, even using the API, before my data finally downloaded.

PR #615 (2018-12-14): ENH: iotools ecmwf macc
pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python! You may delete all of these instructions except for the list below.

You may submit a pull request with your code at any stage of completion.

The following items must be addressed before the code can be merged. Please don't hesitate to ask for help if you're unsure of how to accomplish any of the items below:

 - [x] Closes  #602
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html) [... cut]
closed 2018-12-14 by cwhanse