pvlib/pvlib-python #607: atmosphere.py bug fix on get_relative_airmass airmass

jramirezpr, 2018-10-18:
**Describe the bug**
A clear and concise description of what the bug is.
on conda download, line 222  trails off
   am = (1.0 / (np.cos(zenith_rad) +
**fix**
join with line 223
        am = (1.0 / (np.cos(zenith_rad) + 0.50572*(((6.07995 + (90 - z)) ** - 1.6364))))
 or include \ to denote continuation

mikofski, 2018-10-18:
Hi @jramirezpr , thanks for reporting your issue. 😄 
Can you please execute
```python
import pvlib
pvlib.atmosphere.get_relative_airmass(zenith=60.0, model='kastenyoung1989')
```
and post here either the output or the traceback ?  Also please output `pvlib.__version__` and your version of Python.

I'm using conda (Python-3.7) and the latest version of pvlib (0.6.0+27.g3077923) and I get `1.9942928525292494` and no traceback.

jramirezpr, 2018-10-18:
The error seems to be when zenith>90, since it filters the entry to  NaN and you get a problem with np.cos

cwhanse, 2018-10-18:
@jramirezpr why do you want to calculate air mass when zenith>90?

jramirezpr, 2018-10-18:
I'm running the example on the documentation https://wholmgren-pvlib-python-new.readthedocs.io/en/extras/introexamples.html and some of the zeniths in solpos['apparent_zenith'] are greater than 90

cwhanse, 2018-10-18:
@jramirezpr I'm having trouble duplicating your issue.  Is this the warning you get?

RuntimeWarning: invalid value encountered in maximum
  spectral_loss = np.maximum(0, np.polyval(am_coeff, airmass_absolute))

jramirezpr, 2018-10-18:
@cwhanse it is basically the same reason, (passing a nan) except on line 222.

jramirezpr, 2018-10-18:
@cwhanse  (it is in the get_relative_airmass function,after the zeniths greater than 90 have been filtered with nan)

cwhanse, 2018-10-18:
What version of pvlib?

jramirezpr, 2018-10-18:
@cwhanse  0.6.0-py_1

cwhanse, 2018-10-18:
numpy version?

jramirezpr, 2018-10-18:
@cwhanse numpy 1.15.1

cwhanse, 2018-10-18:
Could you post your code? I can't reproduce that warning.

mikofski, 2018-10-18:
Sorry, I'm also having trouble reproducing your error. I copied and pasted the intro you linked to, and I get the same output as the intro

<details>

```python
In [1]: %paste
In [1]: import pandas as pd

In [2]: import matplotlib.pyplot as plt

In [3]: naive_times = pd.DatetimeIndex(start='2015', end='2016', freq='1h')

# very approximate
# latitude, longitude, name, altitude, timezone
In [4]: coordinates = [(30, -110, 'Tucson', 700, 'Etc/GMT+7'),
   ...:                (35, -105, 'Albuquerque', 1500, 'Etc/GMT+7'),
   ...:                (40, -120, 'San Francisco', 10, 'Etc/GMT+8'),
   ...:                (50, 10, 'Berlin', 34, 'Etc/GMT-1')]
   ...:

In [5]: import pvlib

# get the module and inverter specifications from SAM
In [6]: sandia_modules = pvlib.pvsystem.retrieve_sam('SandiaMod')

In [7]: sapm_inverters = pvlib.pvsystem.retrieve_sam('cecinverter')

In [8]: module = sandia_modules['Canadian_Solar_CS5P_220M___2009_']

In [9]: inverter = sapm_inverters['ABB__MICRO_0_25_I_OUTD_US_208_208V__CEC_2014_']

# specify constant ambient air temp and wind for simplicity
In [10]: temp_air = 20

In [11]: wind_speed = 0

## -- End pasted text --

In [2]: %paste
In [12]: system = {'module': module, 'inverter': inverter,
   ....:           'surface_azimuth': 180}
   ....:

In [13]: energies = {}

In [14]: for latitude, longitude, name, altitude, timezone in coordinates:
   ....:     times = naive_times.tz_localize(timezone)
   ....:     system['surface_tilt'] = latitude
   ....:     solpos = pvlib.solarposition.get_solarposition(times, latitude, longitude)
   ....:     dni_extra = pvlib.irradiance.get_extra_radiation(times)
   ....:     airmass = pvlib.atmosphere.get_relative_airmass(solpos['apparent_zenith'])
   ....:     pressure = pvlib.atmosphere.alt2pres(altitude)
   ....:     am_abs = pvlib.atmosphere.get_absolute_airmass(airmass, pressure)
   ....:     tl = pvlib.clearsky.lookup_linke_turbidity(times, latitude, longitude)
   ....:     cs = pvlib.clearsky.ineichen(solpos['apparent_zenith'], am_abs, tl,
   ....:                                  dni_extra=dni_extra, altitude=altitude)
   ....:     aoi = pvlib.irradiance.aoi(system['surface_tilt'], system['surface_azimuth'],
   ....:                                solpos['apparent_zenith'], solpos['azimuth'])
   ....:     total_irrad = pvlib.irradiance.get_total_irradiance(system['surface_tilt'],
   ....:                                                         system['surface_azimuth'],
   ....:                                                         solpos['apparent_zenith'],
   ....:                                                         solpos['azimuth'],
   ....:                                                         cs['dni'], cs['ghi'], cs['dhi'],
   ....:                                                         dni_extra=dni_extra,
   ....:                                                         model='haydavies')
   ....:     temps = pvlib.pvsystem.sapm_celltemp(total_irrad['poa_global'],
   ....:                                          wind_speed, temp_air)
   ....:     effective_irradiance = pvlib.pvsystem.sapm_effective_irradiance(
   ....:         total_irrad['poa_direct'], total_irrad['poa_diffuse'],
   ....:         am_abs, aoi, module)
   ....:     dc = pvlib.pvsystem.sapm(effective_irradiance, temps['temp_cell'], module)
   ....:     ac = pvlib.pvsystem.snlinverter(dc['v_mp'], dc['p_mp'], inverter)
   ....:     annual_energy = ac.sum()
   ....:     energies[name] = annual_energy
   ....:

In [15]: energies = pd.Series(energies)

# based on the parameters specified above, these are in W*hrs
In [16]: print(energies.round(0))

## -- End pasted text --
pvlib/pvsystem.py:1917: RuntimeWarning: invalid value encountered in maximum
  spectral_loss = np.maximum(0, np.polyval(am_coeff, airmass_absolute))
Tucson           467740.0
Albuquerque      500451.0
San Francisco    440046.0
Berlin           383547.0
dtype: float64
```
</details>

here's my conda info:
```python

In [3]: import numpy as np

In [4]: np.__version__
Out[4]: '1.15.2'

In [5]: pd.__version__
Out[5]: '0.23.4'

In [6]: pvlib.__version__
Out[6]: '0.6.0+27.g3077923'

In [7]: import sys

In [8]: sys.version
Out[8]: '3.7.0 (default, Jun 28 2018, 08:04:48) [MSC v.1912 64 bit (AMD64)]'
```

furthermore if I execute `np.cos(np.nan)` the output is `nan` no trackback
```python
In [2]: np.cos(np.nan)
Out[2]: nan

In [9]: 1 / np.cos(np.nan)
Out[9]: nan
```

So I need more information to try to figure out what is going on with your code. Can you try to debug? If you start IPython, and type `%pdb` then you should see

    Automatic pdb calling has been turned ON

and then when you call `get_relative_airmass` it should drop into the debugger on the line where your exception is raised, then you can output all of the values, execute statements, etc.
```python
In [10]: pvlib.atmosphere.get_relative_airmass(zenith=np.nan, model='kastenyoung1989')
NOTE: Enter 'c' at the ipdb>  prompt to continue execution.
> c:\users\mikm\projects\pvlib-python\pvlib\atmosphere.py(216)get_relative_airmass()
    214     # need to filter first because python 2.7 does not support raising a
    215     # negative number to a negative power.
--> 216     z = np.where(zenith > 90, np.nan, zenith)
    217     zenith_rad = np.radians(z)
    218

ipdb>
```
See the [Python docs on debugging](https://docs.python.org/3/library/pdb.html). Any info you can find out will help us trouble shoot this from our end. Thanks!

jramirezpr, 2018-10-18:
@mikofski @cwhanse sure, i'll use the debugger .shown is a printscr with the warning.
[image]
And here is the copypasta code:

# for a given hardware configuration at a handful of sites listed below. 
import pandas as pd
import matplotlib.pyplot as plt 
import numpy as np 
 
 #dataframe of time intervals
naive_times = pd.DatetimeIndex(start='2015', end='2016', freq='1h')
#coordinates will be list of sites for instalation, each entry  
#is a tuple with the following values
# (latitude, longitude, name, altitude, timezone)
coordinates = [(30, -110, 'Tucson', 700, 'Etc/GMT+7'),  (35, -105, 'Albuquerque', 1500, 'Etc/GMT+7'),]
#modules from SAM

import pvlib
sandia_modules = pvlib.pvsystem.retrieve_sam('SandiaMod')
sapm_inverters = pvlib.pvsystem.retrieve_sam('cecinverter')
#module and inverter example
module = sandia_modules['Canadian_Solar_CS5P_220M___2009_']
inverter = sapm_inverters['ABB__MICRO_0_25_I_OUTD_US_208_208V__CEC_2014_']
 
 #IMPORTANT NOTE: specify constant ambient air temp and wind for simplicity
# MAYBE USE WEATHER FORECASTS LATER???
 
temp_air = 20
wind_speed = 0
#dictionary of system installed. these are the constant features,
# other features will be modified in the loop according to site info.
system = {'module': module, 'inverter': inverter,'surface_azimuth':180}
#dictionary of energies
energies = {}
#add predicted energy to each site
for latitude, longitude, name, altitude, timezone in coordinates:
    times = naive_times.tz_localize(timezone)
    system['surface_tilt'] = latitude
    solpos = pvlib.solarposition.get_solarposition(times, latitude, longitude)
    dni_extra = pvlib.irradiance.get_extra_radiation(times)
    
    #note,relative airmass has warnings so maybe bugs?
    airmass = pvlib.atmosphere.get_relative_airmass(solpos['apparent_zenith'])
    pressure = pvlib.atmosphere.alt2pres(altitude)  
    am_abs = pvlib.atmosphere.get_absolute_airmass(airmass, pressure)
    tl = pvlib.clearsky.lookup_linke_turbidity(times, latitude, longitude)
    cs = pvlib.clearsky.ineichen(solpos['apparent_zenith'], am_abs, tl,\
                                 dni_extra=dni_extra, altitude=altitude)
    aoi = pvlib.irradiance.aoi(system['surface_tilt'], system['surface_azimuth'],
                               solpos['apparent_zenith'], solpos['azimuth'])
    total_irrad = pvlib.irradiance.get_total_irradiance(system['surface_tilt'],
                                                        system['surface_azimuth'],
                                                        solpos['apparent_zenith'],
                                                        solpos['azimuth'],
                                                        cs['dni'], cs['ghi'],dhi= cs['dhi'],
                                                        dni_extra=dni_extra,
                                                        model='haydavies')
    #from here on, ENERGY for system is calculated
    temps = pvlib.pvsystem.sapm_celltemp(total_irrad['poa_global'], wind_speed, temp_air)

mikofski, 2018-10-18:
Sorry I still can't repeat your issue.

Can you exit Spyder, start the Anaconda Prompt, and execute `conda list?

<details>
<summary>Details: <b>click to expand!</b></summary>

```
(py37) C:\Users\mikm>conda list
# packages in environment at C:\Users\mikm\AppData\Local\Continuum\miniconda3\envs\py37:
#
# Name                    Version                   Build  Channel
asn1crypto                0.24.0                   py37_0
astroid                   2.0.4                    py37_0
atomicwrites              1.2.1                    py37_0
attrs                     18.2.0           py37h28b3542_0
backcall                  0.1.0                    py37_0
beautifulsoup4            4.6.3                    py37_0
blas                      1.0                         mkl
blosc                     1.14.4               he51fdeb_0
bzip2                     1.0.6                hfa6e2cd_5
ca-certificates           2018.03.07                    0
certifi                   2018.10.15               py37_0
cffi                      1.11.5           py37h74b6da3_1
cftime                    1.0.0b1          py37h452e1ab_0
chardet                   3.0.4                    py37_1
colorama                  0.3.9                    py37_0
cryptography              2.3.1            py37h74b6da3_0
curl                      7.61.1               h7602738_0
cycler                    0.10.0                   py37_0
cython                    0.28.5           py37h6538335_0
decorator                 4.3.0                    py37_0
ephem                     3.7.6.0          py37hfa6e2cd_1
freetype                  2.9.1                ha9979f8_1
hdf4                      4.2.13               h712560f_2
hdf5                      1.10.2               hac2f561_1
icc_rt                    2017.0.4             h97af966_0
icu                       58.2                 ha66f8fd_1
idna                      2.7                      py37_0
intel-openmp              2019.0                      118
ipython                   7.0.1            py37h39e3cac_0
ipython_genutils          0.2.0                    py37_0
isort                     4.3.4                    py37_0
jedi                      0.12.1                   py37_0
jinja2                    2.10                     py37_0
jpeg                      9b                   hb83a4c4_2
kiwisolver                1.0.1            py37h6538335_0
lazy-object-proxy         1.3.1            py37hfa6e2cd_2
libcurl                   7.61.1               h7602738_0
libnetcdf                 4.6.1                h70ac11d_1
libpng                    1.6.34               h79bbb47_0
libssh2                   1.8.0                hd619d38_4
llvmlite                  0.25.0                   py37_0
lzo                       2.10                 h6df0209_2
markupsafe                1.0              py37hfa6e2cd_1
matplotlib                2.2.3            py37hd159220_0
mccabe                    0.6.1                    py37_1
mkl                       2019.0                      118
mkl_fft                   1.0.6            py37hdbbee80_0
mkl_random                1.0.1            py37h77b88f5_1
more-itertools            4.3.0                    py37_0
netcdf4                   1.4.1            py37hbfe741f_0
numba                     0.40.0           py37hf9181ef_0
numexpr                   2.6.8            py37h9ef55f4_0
numpy                     1.15.2           py37ha559c80_0
numpy-base                1.15.2           py37h8128ebf_0
openssl                   1.0.2p               hfa6e2cd_0
pandas                    0.23.4           py37h830ac7b_0
parso                     0.3.1                    py37_0
pickleshare               0.7.5                    py37_0
[... 19 lines cut ...]
scipy                     1.1.0            py37h4f6bf74_1
setuptools                40.4.3                   py37_0
simplegeneric             0.8.1                    py37_2
sip                       4.19.12          py37h6538335_0
siphon                    0.8.0                     <pip>
six                       1.11.0                   py37_1
snappy                    1.1.7                h777316e_3
sqlite                    3.25.2               hfa6e2cd_0
tornado                   5.1.1            py37hfa6e2cd_0
traitlets                 4.3.2                    py37_0
urllib3                   1.23                     py37_0
vc                        14.1                 h0510ff6_4
vs2015_runtime            14.15.26706          h3a45250_0
wcwidth                   0.1.7                    py37_0
wheel                     0.32.0                   py37_0
win_inet_pton             1.0.1                    py37_1
wincertstore              0.2                      py37_0
wrapt                     1.10.11          py37hfa6e2cd_2
zlib                      1.2.11               h8395fce_2
```
</details><br/>

How did you install pvlib? Did you use pip? conda forge? or the pvlib conda channel?
Did you create  a new conda environment or use the base anaconda environment?
Are you using Anaconda or Miniconda?

jramirezpr, 2018-10-19:
I installed with anaconda prompt , ran the
conda conda -n pvlib command.
I am using anaconda, through the navigator, and spyder as you mentioned

Details: click to expand!

mikofski, 2018-10-19:
Can you please be more specific? Starting from scratch, what were the exact steps you took to install and run pvlib? Thanks.

wholmgren, 2018-10-19:
numpy warnings can be very sensitive to platform, python version, numpy version, and input type. I think it's good practice for pvlib to avoid spitting warnings for typical use cases, so I'd welcome a pull request to strategically avoid or silence the warning.

@jramirezpr you can also call the pvlib function within a [np.errstate](https://docs.scipy.org/doc/numpy-1.15.0/reference/generated/numpy.errstate.html) block to silence the warning.

mikofski, 2018-10-19:
I vote not to silence it, let the user decide to silence it. IMO pvlib is a toolbox not production, so it should raise all warnings and exceptions

jramirezpr, 2018-10-19:
I think I will do that, I guess there is no airmass formula for nighttime because the function is mostly auxiliary for the rest of PV analysis and so it doesn’t expect those values to be inputted?

wholmgren, 2018-10-19:
> I vote not to silence it, let the user decide to silence it. IMO pvlib is a toolbox not production, so it should raise all warnings and exceptions

@mikofski I agree with the spirit of this but I in practice I think it's more complicated. numpy warnings are often emitted for valid inputs due to well-understood implementation details. log(0), exp(very big number), cos(nan), etc. I think pvlib should implement things in a way that the warnings are not emitted or selectively silence them in `with` blocks where necessary. This way users are more likely to think twice when they do see a warning. That's the core of this issue, at least so far as I understand it.

> I guess there is no airmass formula for nighttime because the function is mostly auxiliary for the rest of PV analysis and so it doesn’t expect those values to be inputted?

Airmass is not defined when zenith is greater than ~90 degrees, so the function returns nan. The function emits this warning because it sets the nans before the calculating the airmass.

jramirezpr, 2018-10-22:
@wholmgren  so should it be better to filter out the times for which the zenith is out of range? Or does it not really matter?

wholmgren, 2018-10-22:
@jramirezpr here are some examples of how to avoid similar numpy warnings:

use out and where kwarg in calculation: https://github.com/pvlib/pvlib-python/blob/master/pvlib/pvsystem.py#L916-L920

use the errstate context manager: https://github.com/pvlib/pvlib-python/blob/master/pvlib/pvsystem.py#L2658

let us know if one of these approaches works for you

jramirezpr, 2018-10-23:
@wholmgren Thank you, I'll try this out.

jramirezpr, 2018-10-23:
@cwhanse on a similar note, i found some more warnings when changing the model to perez
C:\Users\Guest\Anaconda3\lib\site-packages\pvlib\irradiance.py:1150: RuntimeWarning: invalid value encountered in maximum
  F1 = np.maximum(F1, 0)
C:\Users\Guest\Anaconda3\lib\site-packages\pvlib\irradiance.py:1153: RuntimeWarning: invalid value encountered in maximum
  F2 = np.maximum(F2, 0)
C:\Users\Guest\Anaconda3\lib\site-packages\pvlib\irradiance.py:1167: RuntimeWarning: invalid value encountered in maximum
  sky_diffuse = np.maximum(dhi * (term1 + term2 + term3), 0)

[image]
I'm going to silence them for now, but thought you might want to know.

cwhanse, 2018-10-23:
@jramirezpr thank you , appreciate this!

mikofski, 2018-10-23:
can this issue be closed now?

jramirezpr, 2018-10-23:
I have no problem with closing it

cwhanse, 2018-10-23:
Hold it for a bit while I look at the F1, F2 warnings @jramirezpr posted.  I'll close.

cwhanse, 2018-10-23:
Warnings are from impolite handling of `np.nan` in the `perez` function.

closed 2018-10-23 by cwhanse