pvlib/pvlib-python #630: atol parameter too wide for assert_allclose() in test_pvsystem.py

CameronTStark, 2018-12-07:
**Describe the bug**
There are two instances in the test_pvsystem.py (lines 246 & 1198) where numpy.testing.assert_allclose() has values listed under 'atol' that create an "absolute tolerance" that are too big to catch changes. 

**Additional context**
I noticed this by chance when writing the tests for pvsystem.pvsyst_celltemp(). It's possible that these functions/tests have been problematic before and these wide error bars were provided to avoid "nuisance alarms". I have a PR lined up for this with just the changes on those two lines and can flesh out the rest of the PR (e.g. documentation)  if this is indeed an issue.

cwhanse, 2018-12-07:
Do you mean line 1396 rather than 1198?

For line 246, that test is only confirming that the function `pvsystem.sapm` is called when the method `PVSystem.sapm` is invoked.  Similar purpose for the test that includes line 1396.  Could use a comment, or perhaps a different test condition, to make the purpose more clear.  @wholmgren ?

wholmgren, 2018-12-07:
@cwhanse is correct. Those tests only assert that the function was called and that it returned a value that was reasonable. The tests for the pure functions contain the stringent tolerance requirements.

>It's possible that these functions/tests have been problematic before and these wide error bars were provided to avoid "nuisance alarms".

exactly.

CameronTStark, 2018-12-07:
Good deal, thought I'd check, have a nice weekend!

closed 2018-12-11 by CameronTStark