pvlib/pvlib-python #637: _delta_kt_prime_dirint() Method Returns Values That Are Half of What they Should be

chriswmackey, 2019-01-03:
Hello,

I believe that I have found a bug or at least a misalignment between a formula that was written in a publication and the result of a pvlib method.

The bug appears specifically on these lines of the _delta_kt_prime_dirint() method in irradiance.py:
https://github.com/pvlib/pvlib-python/blob/master/pvlib/irradiance.py#L1566-L1568

... and I realize that this formula references Perez's equation 2 as follows:
[image]

Here is the full publication where the image above came from from:
https://www.researchgate.net/publication/279868352_Dynamic_global-to-direct_irradiance_conversion_models

The bug seems to be the result of how pandas deals with the `add()` method and the `fill_value`.

To re-create the bug, I recommend running the following code:
```
kt_prime = pd.Series([0.79, 0.77])
times = pd.DatetimeIndex(['2014-06-24T12-0700', '2014-06-24T18-0700'])
delta_kt_prime = pvlib.irradiance._delta_kt_prime_dirint(kt_prime, True, times)
```

At the moment, this returns a delta_kt_prime value that is equal to 0.01.  However, this should clearly be equal to 0.02 according to Perez's equation 2.

**Versions:**
 - ``pvlib.__version__``: 0.6.0
 - ``pandas.__version__``:  0.23.4
 - python: 3.7

cwhanse, 2019-01-03:
@chriswmackey verified

cwhanse, 2019-01-03:
@chriswmackey thanks very much for bringing this to our attention!

cwhanse, 2019-01-03:
This bug affects the first and last values in a time series of DNI calculated by `irradiance.dirint`.  The error is not "small", e.g., with GHI = 1038 W/m2 and zenith = 10.6 degrees, the incorrect value of DNI was 888, after correction DHI is 868.

closed 2019-01-09 by cwhanse

wholmgren, 2019-01-09:
thanks @chriswmackey for the report and @cwhanse for the fix.

PR #638 (2019-01-09): fix _delta_kt_prime_dirint end values
pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python! You may delete all of these instructions except for the list below.

You may submit a pull request with your code at any stage of completion.

The following items must be addressed before the code can be merged. Please don't hesitate to ask for help if you're unsure of how to accomplish any of the items below:

 - [x] Closes  #637
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/cont [... cut]