pvlib/pvlib-python #639: Use of greek letters and possibly other symbols in python identifiers

adriesse, 2019-01-05:
Naming things has been deemed a hard problem by smarter people than me, but python's ability to use greek lettters (among other things) as (part of) identifiers seems to open up new possibilities.  Is anyone using this in their own code?  And is this something that could be done in pvlib-python?  Does github allow it?

markcampanelli, 2019-01-06:
My understanding is that this requires Python3's unicode support for Python identifiers.

adriesse, 2019-01-08:
That is true.

wholmgren, 2019-01-09:
I support using greek letters in documentation and especially notebooks. I'm not entirely opposed to using them in the code once we move to Python 3, but we'd need to explore the pros/cons on our  development platforms.

adriesse, 2019-01-11:
Would this be a good place to accumulate observations?  
I will try it in my upcoming work and will limit myself to a fairly small subset.

wholmgren, 2019-01-11:
> Would this be a good place to accumulate observations?

Yes, works for me.