pvlib/pvlib-python #641: Negative "POA horizontal ratio" when sun behind array

adriesse, 2019-01-11:
**Describe the bug**
The function `poa_horizontal_ratio` gives negative values when the sun is behind the array.

**To Reproduce**
pvlib.irradiance.poa_horizontal_ratio(60, 90, 60, 270)
-0.99999999999999933

**Expected behavior**
pvlib.irradiance.poa_horizontal_ratio(60, 90, 60, 270)
0.0

**Versions:**
 - ``pvlib.__version__``: '0.5.2' 
 - ``pandas.__version__``: '0.19.2'
 - python: 3.5.1

**Additional context**
Similar problems were fixed in other irradiance functions already (#526).

wholmgren, 2019-01-12:
I think we considered this behavior in #535 and decided against limits in this function. https://github.com/pvlib/pvlib-python/pull/535/commits/9a74404808d122b4c55b9292a2cf6a24eeb17039

adriesse, 2019-01-12:
Sorry, I guess I did not look in the right place for the latest code.  I looked here:

[https://pvlib-python.readthedocs.io/en/latest/_modules/pvlib/irradiance.html](https://pvlib-python.readthedocs.io/en/latest/_modules/pvlib/irradiance.html)

wholmgren, 2019-01-14:
@adriesse would you suggest that we change the behavior or at least the documentation? If so, happy to keep the issue open.

adriesse, 2019-01-14:
Slightly changing both would be best in my view.  Allowing `aoi_projection` to go negative is ok, but the docstring could be expanded to explain the meaning of negative values. I think that `poa_horizontal_ratio` should only use non-negative values of `aoi_projection` like the other users of that function.  But wouldn't put a high priority on it--I just noticed it as I was browsing the file looking for something else.

kandersolar, 2024-10-19:
`poa_horizonal_ratio` was removed in #2021.  Shall we close this issue?

closed 2024-10-20 by adriesse