pvlib/pvlib-python #643: rewrite forecast module to get data from NCEP

wholmgren, 2019-01-18:
**Is your feature request related to a problem? Please describe.**
The `forecast.py` module provides convenient access to cloud cover and irradiance data from weather models stored on a Unidata THREDDS server. But sometimes I want model data that's not available from Unidata THREDDS. For example, HRRR subhourly irradiance data is available in NCEP's grib files, but not from Unidata. As another example, the period of record on the Unidata THREDDS server is typically shorter than the online NCEP servers.

**Describe the solution you'd like**
Rewrite `forecast.py` to get data directly from NCEP. I think this is more feasible than it used to be thanks to recent improvements in `xarray`.

**Describe alternatives you've considered**
See if Unidata/NOAA can add more variables to the THREDDS service.

**Additional context**
This supports the Solar Forecasting 2 projects.

meteoDaniel, 2019-02-12:
I suggest to implement `grib_filter` API. 
`https://nomads.ncep.noaa.gov/cgi-bin/filter_gfs_0p25_1hr.pl?file=gfs.t12z.pgrb2.0p25.anl&var_ALBDO=on&var_DLWRF=on&var_DSWRF=on&subregion=&leftlon=0&rightlon=360&toplat=90&bottomlat=-90&dir=%2Fgfs.2019020212`

MorningGlory747, 2019-03-05:
From what I've seen on the matter, the grid_filter API gives out a .grib format. To manipulate the file into giving point data, such as what pvlib forecast currently does, it would be necessary to use a module that can operate on grib files, such as pygrib. However, to my knowledge, using pygrib on a windows is a bit complicated and could prove itself to be a frustrating experience for some users. 

I am curious to know what is the current state of progress on this matter, as I have found myself to be limited by the amount of different weather variables available on Unidata THREDDS. Have there been any pertinent improvements since the initial posting date?

wholmgren, 2019-05-03:
For the near (~1 year) future, all major forecast related activity will be in https://github.com/SolarArbiter/solarforecastarbiter-core. See, for example, the [nwp fetch script](https://github.com/SolarArbiter/solarforecastarbiter-core/blob/master/solarforecastarbiter/io/fetch/nwp.py). In that project, we are using the grib filter API to get data from NCEP. Supporting that or similar code in pvlib would be a major distraction from its core mission, so I'm going to close this issue.

closed 2019-05-03 by wholmgren