pvlib/pvlib-python #646: scalar inputs yield ndim=0 array outputs

wholmgren, 2019-01-19:
**Describe the bug**
pvlib has a policy that the output type should be as similar as possible to the input type. So, for all scalar input, most functions should return a scalar output. However, some functions return ndim=0 array outputs instead of true scalars. I know this happens with `irradiance.perez`, but I think it happens with a handful of functions. 

**To Reproduce**
```python
    out = irradiance.perez(40, 180, 118.45831879, 939.95469881,
                           1321.1655834833093, 10.56413562, 144.76567754,
                           1.01688136)
    assert np.isscalar(out)
```

**Expected behavior**
The function should return a scalar, not an ndim=0 array.

**Versions:**
 - ``pvlib.__version__``: 0.6.0
 - ``pandas.__version__``: many
- ``numpy.__version__``: many
 - python: many

**Additional context**
We've previously discussed creating utility functions and decorators to handle type issues like this one. See #635 for example. 

The first thing we need to do is catalog which functions exhibit this behavior.

markcampanelli, 2019-01-19:
This is a bit of a PITA: https://github.com/numpy/numpy/issues/12636

mikofski, 2019-01-19:
[I recently got bit by this too](https://github.com/scipy/scipy/pull/9612)

markcampanelli, 2019-01-19:
@mikofski Thanks for pushing on that vectorized newton in scipy, it has been quite useful. I am a bit concerned about the different output formats in the scalar vs. vectorized cases, however. Seems like that leads to the same issue we are discussing here.

wholmgren, 2019-01-21:
Thanks @markcampanelli for raising the issue with numpy and bringing it to my attention here. I'm less convinced that we should put much effort into it within pvlib.