pvlib/pvlib-python #649: KeyError: 'ECMWF_API_KEY' when importing pvlib

adriesse, 2019-01-19:
**Describe the bug**

I have a clean new Anaconda install including the ecmwf-api-client, but my `.ecmwfapirc` was not in my home or working directory.  This caused `import pvlib` to fail badly.

**To Reproduce**

1. import pvlib
2. See error 

**Screenshots**

> Traceback (most recent call last):
>   File "C:\Anaconda3\envs\p37-dev\lib\site-packages\ecmwfapi\api.py", line 54, in _get_apikey_from_environ
>     key = os.environ["ECMWF_API_KEY"]
>   File "C:\Anaconda3\envs\p37-dev\lib\os.py", line 678, in __getitem__
>     raise KeyError(key) from None
> KeyError: 'ECMWF_API_KEY'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "C:\Anaconda3\envs\p37-dev\lib\site-packages\ecmwfapi\api.py", line 97, in get_apikey_values
>     key_values = _get_apikey_from_environ()
>   File "C:\Anaconda3\envs\p37-dev\lib\site-packages\ecmwfapi\api.py", line 59, in _get_apikey_from_environ
>     raise APIKeyFetchError("ERROR: Could not get the API key from the environment")
> ecmwfapi.api.APIKeyFetchError: ERROR: Could not get the API key from the environment
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "C:\Anaconda3\envs\p37-dev\lib\site-packages\ecmwfapi\api.py", line 66, in _get_apikey_from_rcfile
>     with open(rc) as f:
> FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\anton\\.ecmwfapirc'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
>   File "d:\pvcode\lib\pvlib-python-adriesse\pvlib-python\pvlib\__init__.py", line 9, in <module>
>     from pvlib import iotools
>   File "d:\pvcode\lib\pvlib-python-adriesse\pvlib-python\pvlib\iotools\__init__.py", line 8, in <module>
>     from pvlib.iotools.ecmwf_macc import read_ecmwf_macc  # noqa: F401
>   File "d:\pvcode\lib\pvlib-python-adriesse\pvlib-python\pvlib\iotools\ecmwf_macc.py", line 28, in <module>
>     SERVER = ECMWFDataServer()
>   File "C:\Anaconda3\envs\p37-dev\lib\site-packages\ecmwfapi\api.py", line 483, in __init__
>     key, url, email = get_apikey_values()
>   File "C:\Anaconda3\envs\p37-dev\lib\site-packages\ecmwfapi\api.py", line 100, in get_apikey_values
>     key_values = _get_apikey_from_rcfile()
>   File "C:\Anaconda3\envs\p37-dev\lib\site-packages\ecmwfapi\api.py", line 69, in _get_apikey_from_rcfile
>     raise APIKeyFetchError(str(e))
> ecmwfapi.api.APIKeyFetchError: [Errno 2] No such file or directory: 'C:\\Users\\anton\\.ecmwfapirc' 

**Versions:**
 - pvlib.__version__: 6.1-beta
 - python: 3.7

**Additional context:**
New relevant code was introduced in #602.

mikofski, 2019-01-19:
Hi Anton,

Thanks for this report. Bummer. I'm sorry for the inconveniences.

So I think this is an issue with the ecmwf-api-client, not pvlib-python. And I think this is the expected behavior, although maybe it should be documented better. [Step 4 in the "Access ECMWF public datasets"](https://confluence.ecmwf.int/plugins/servlet/mobile?contentId=23694554#content/view/47293609) says to place the key in the `$HOME` folder.  I assume that if the key isn't in the `$HOME` folder then the environment variable `ECMWF_API_KEY` should be set to its location. However there isn't any mention of this in the documentation.

I suggest clarifying the contract with the user more explicitly in the pvlib-python docstring, and asking ECMWF about the `ECMWF_API_KEY` environment variable.

adriesse, 2019-01-19:
It was no big deal--it didn't take long to figure out.  My impression is that you shouldn't do `SERVER = ECMWFDataServer()` during the import, but only later when/if you need it.

mikofski, 2019-01-19:
Thanks for the feedback. [`SERVER = ECMWFDataServer()`](https://github.com/pvlib/pvlib-python/blob/master/pvlib/iotools/ecmwf_macc.py#L18) is in a `try` block that only gets set if no exception is raised. I think you only get the exception because the ecmwf-api-client is installed but the `ECMWF_API_KEY is misconfigured

wholmgren, 2019-01-21:
I don't think pvlib import should be contingent on the proper configuration of `ECMWF_API_KEY` (regardless of whether or not `ecmwf-api-client` is installed). I suggest that we fix this before we release the new feature.

wholmgren, 2019-01-22:
closed by #650

closed 2019-01-22 by wholmgren