pvlib/pvlib-python #656: singleaxis tracking: backtracking error

cwhanse, 2019-02-04:
**Describe the bug**
Low sun angles with a tilted tracking axis cause an error in the backtracking calculation. Code attempts arcos() of a value outside [-1, 1]

**To Reproduce**
```
from pvlib.tracking import singleaxis
result = singleaxis(apparent_zenith=80, apparent_azimuth=338, axis_tilt=30,
                    axis_azimuth=180, max_angle=60, backtrack=True, gcr=0.35)
```
**Expected behavior**
tracker angle should set to max_angle

**Versions:**
 - ``pvlib.__version__``: 0.6.1
 - ``pandas.__version__``:  0.23.3
 - python: 3.6

kevinsa5, 2019-02-07:
I admit I have not read the reference paper, nor do I fully understand the algorithm implementation, but I believe the snippet of interest is this: https://github.com/pvlib/pvlib-python/blob/d6213276f63969ab87c0a154cb947152156a6aa6/pvlib/tracking.py#L422-L431

Changing `temp` to be constrained on both sides appears to solve the issue:
`temp = np.clip(axes_distance*cosd(wid), -1, 1)`

closed 2019-05-03 by wholmgren

wholmgren, 2019-11-20:
If I remember correctly, these inputs produced `nan` before the fix in #697. It's not obvious to me that the expected outcome for these inputs should be max_angle. Maybe the expected outcome in the original post should have stated "tracker angle should be between 0 and +/- max_angle". Or maybe the fix was incomplete. @mikofski do you think the output is wrong?

mikofski, 2019-11-21:
TIL that when the tracker axes are tilted, it's possible for the sun to come from below the trackers, in their reference frame, something I never even considered until I started looking into these low sun angles.

>mikofski do you think the output is wrong?

Yes, I believe it may be wrong. I believe the tracker does not need to backtrack, but it is rotating past the max angle of 90[deg] so it's rotation should be 90[deg], and the AOI calculated from that.

I don't think it needs to backtrack because if you turn off backtracking and set the max angle to 180[deg], then it yields:

```python
{'tracker_theta': array([129.68948615]),
 'aoi': array([61.35300178]),
 'surface_azimuth': array([292.53615425]),
 'surface_tilt': array([56.42233095])}
```

I think the tracker axis tilt and low sun angle have combined to put the sun below the tracker. I don't think it shades the next tracker, but it has exceeded the max angle.

[image]

If the sun is below the tracker then the shadow of the tracker is cast upwards and would shade the top of the next tracker if the shadow's length `LR = L/cos(pi-R)` were greater than the pitch between rows, `P = L/GCR`, so `LR > P` is the same as `cos(pi-R) < GCR`, But `cos(pi-R) = -cos(R) = -cos(129.7) = 0.64 > 0.35` so no backtracking.

I put together a [gist](https://gist.github.com/mikofski/690fc526b0af1d42f48f2a883cee5fd4) to explain what I think is going on. Basically when R is grater than 90[deg] then the condition for backtracking changes because the sun is now below the trackers, and the angle to calculate LR is now `pi - R`, otherwise the length is negative.

The patch applied in #697 also sees `cos(R)/gcr < 1` but the `np.clip` sets the backtracking to 180 because `np.arccos(-1) = 180`.

IMHO the trackers would not backtrack all the way to the east, unless this was a special bifacial tracking solution. For monofacial trackers I think they would stop at the stow position.

mikofski, 2019-11-21:
Moreover, I think this AOI is for the backside that is still facing west.

cwhanse, 2019-11-21:
> that when the tracker axes are tilted, it's possible for the sun to come from below the trackers

It's possible that this happens under very restricted conditons: the sun has to be behind _all_ possible planes to which the modules could be rotated, which would restrict the sun to a relatively narrow "triangle" of azimuth (base) and elevation (height) centered at north azimuth (for a north-south tracker). Do you think we've got a case where this happens?

mikofski, 2019-11-21:
Yes, I think this particular case of the 30deg tilted tracker and the low sun angle leads to the rays coming from below the trackers in their reference frame. If you check the solar angle in the yz place it's 10deg, but the tracker is tilted 30deg. If you rotate the solar vector into the tracker reference frame you get the sun in x'z' plane coming in 39.7deg below the tracker. That's why your get a tracker rotation of 129.7deg (=90+39.7) if you turn off backtracking and set max angle to 180deg.

PR #697 (2019-05-03): better nan handling in singleaxis
 - [x] Closes  #622 Closes #656
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style is sufficient. Passes L [... cut]