pvlib/pvlib-python #657: calcparam_desoto fails with cryptic error when weather.temp_air has dtype=object

jkfm, 2019-02-06:
Hi,

I had the following error and it took a while for me to find out the cause:

>  File "<ipython-input-16-ac56e55f49ee>", line 1, in <module>
>     mc.run_model(times=time_mc, weather=weather)
>   File "/home/johanneskaufmann/.virtualenvs/pfp-gUXDy98y/lib/python3.6/site-packages/pvlib/modelchain.py", line 848, in run_model
>     self.dc_model()
>   File "/home/johanneskaufmann/.virtualenvs/pfp-gUXDy98y/lib/python3.6/site-packages/pvlib/modelchain.py", line 409, in singlediode
>     self.temps['temp_cell']))
>   File "/home/johanneskaufmann/.virtualenvs/pfp-gUXDy98y/lib/python3.6/site-packages/pvlib/pvsystem.py", line 307, in calcparams_desoto
>     self.module_parameters['dEgdT'], **kwargs)
>   File "/home/johanneskaufmann/.virtualenvs/pfp-gUXDy98y/lib/python3.6/site-packages/pvlib/pvsystem.py", line 1066, in calcparams_desoto
>     (np.exp(EgRef / (k*(Tref_K)) - (E_g / (k*(Tcell_K))))))
> AttributeError: 'float' object has no attribute 'exp'
> 

I used temperature data which I input from some csv files using pd.read_csv, looking like this:

> weather.temp_air.head()
> Out[18]: 
> 2017-01-03 00:00:00+09:00    10.292
> 2017-01-03 00:30:00+09:00    10.124
> 2017-01-03 01:00:00+09:00     9.987
> 2017-01-03 01:30:00+09:00     9.856
> 2017-01-03 02:00:00+09:00     9.748
> Name: temp_air, dtype: object

After some searching, I found out that the error occurs, because the Series has the attribute `dtype: object` instead of float. Consequentially, the error disappears when I performed 
`weather = weather.astype(float)`

I am a bit unsure, if this constitutes a bug. I feel that to a non-technical user, this behaviour comes unexpected and is a bit confusing, because the weather data looks and feels valid. 

What do you think? Should the method perform an explicit type conversion and/or provide a more detailed/customized error message.

wholmgren, 2019-02-06:
Perhaps the same problem as https://stackoverflow.com/questions/54484478/attributeerror-in-sapm-celltemp-function-when-using-15-minute-weather-data-res#comment95852619_54484478

I would not call this a pvlib bug. Did pandas or numpy change something recently or has this always been the behavior? It's hit multiple users so I'm open to a `ModelChain.run_model` warning if `weather` contains object `dtypes`. 

I don't think we should push type checking down to the low-level functions, nor do I think we should convert types.

jkfm, 2019-02-13:
Thanks for the input! I'll see if I get around creating a PR to warn users.

adriesse, 2019-02-15:
> (np.exp(EgRef / (k(Tref_K)) - (E_g / (k(Tcell_K))))))
> AttributeError: 'float' object has no attribute 'exp'

I puzzled over this bit when the stackoverflow question first came up.  If the error message is accurate, it suggests the the name `np` has been reassigned inadvertently somewhere..

wholmgren, 2019-02-15:
> If the error message is accurate

The error message is misleading - the exception occurs within the line rather than at the line (if that makes any sense). Related issues have been reported on the numpy issue tracker and stackoverflow for many years.

cwhanse, 2019-03-29:
@jkfm any objections to closing this as a pvlib issue?

jkfm, 2019-03-31:
No objections! :)

closed 2019-03-31 by jkfm