pvlib/pvlib-python #665: Cannot find the mentionned source in pvlib.solarposition.ephemeris

MorningGlory747, 2019-02-18:
Greetings,

I am interested in using pvlib.solarposition.ephemeris as it contains a clear set of equations to compute solar radiation. This can eazily be documented compared to other programs such as pyephem which are much harder to track. After testing numerous methods, I consider ephemeris to be a decent artificial solar radiation function albeit its warnings. 

I would like however to find the mentionned reference  : Grover Hughes' class and related class materials on Engineering  Astronomy at Sandia National Laboratories, 1985. I am unable to locate this specific reference, but a document of 1981 (https://www.osti.gov/servlets/purl/6377969) mentions a method/program develop by Grover Hughes called the SUNAE program. The document reveals the equations of the SUNAE program from appendix E to G. However, these equations dont exactly ressemble to what is being used in the current ephemeris code so I was wondering if there would be any way for me to find a specific source which shows a very similar set of equations to what is currently used in pvlib.solarposition.ephemeris?

Thank you,

pvlib usage questions can be asked on [Stack Overflow](http://stackoverflow.com) and tagged with the [pvlib](http://stackoverflow.com/questions/tagged/pvlib) tag.

The [pvlib-python google group](https://groups.google.com/forum/#!forum/pvlib-python) is used for discussing various topics of interest to the pvlib-python community. We also make new version announcements on the google group.

cwhanse, 2019-02-19:
I have a copy of the source document, which we ought to post to pvpmc.org.  Its not a formal publication. It will be a few days before I am at my desk and can post the file.

MorningGlory747, 2019-02-19:
> 
> 
> I have a copy of the source document, which we ought to post to pvpmc.org. Its not a formal publication. It will be a few days before I am at my desk and can post the file.

I am confident that it will suffice nonetheless. Thank you for taking the time in posting it online!

cwhanse, 2019-02-20:
The source for the Sandia ephemeris function is [posted](https://pvpmc.sandia.gov/modeling-steps/1-weather-design-inputs/sun-position/sandias-code/)

closed 2019-02-20 by cwhanse