pvlib/pvlib-python #668: NOAA NCDC global summary of the day GSOD for rainfall in iotools

mikofski, 2019-03-01:
[NREL NCDC database of GSOD](https://data.noaa.gov/dataset/dataset/global-surface-summary-of-the-day-gsod) data has daily rainfall that is useful for determining soling rates based on Addie Kimber's dynamic soiling model. There used to be an ftp site, but I didn't look too hard for it.

Is there a chance that some one from SunPower, like @cedricleroy or @anomam , could dig through the pvlife github repo and extract the gsod code and add it to iotools - or post it to a [gist](https://gist.github.com) so that someone else can work it into shape for `pvlib.iotools`?

thanks!

cwhanse, 2019-03-01:
Nit - that's  NOAA database, not NREL.  The link is correct.

wholmgren, 2019-03-01:
If it's already useful for PV modelers that's fine with me. I am not picky about what goes into `iotools` but I don't want it to be a dumping ground for random weather data. Unless I missed something GSOD does not have irradiance or aerosol data, so it might be better placed in a package like [Unidata's Siphon](https://github.com/Unidata/siphon)

mikofski, 2019-03-01:
I only used it for daily rainfall in the soiling model. If there's a better source of rainfall data, then this probably isn't needed. If it's not useful for pvlib, I can open an issue for it at Siphon, if it's not already there.

mikofski, 2019-03-01:
I think the [noaa ncdc isd](https://www.ncdc.noaa.gov/isd) also has global rainfall

AdamRJensen, 2023-11-22:
@kandersolar With the addition of #1767, is the GSOD dataset mentioned here still desirable? I'm unsure if they retrieve data from the same source.

Link to GSOD: https://www.ncei.noaa.gov/access/metadata/landing-page/bin/iso?id=gov.noaa.ncdc:C00516

kandersolar, 2023-11-22:
I browsed through the list of available stations for Denmark as an example and saw only partial overlap with what ACIS makes available, so I think #1767 does not make this issue redundant.