pvlib/pvlib-python #672: modeling multijunction cells with pvlib

Piranias, 2019-03-14:
I'm interested in using the pvlib to model multijunction CPV cells. I aware that the standard version of SAPM was only developed for flat plate modules. 
There has been an initiative for a Semprius CPV-cell to model the strong spectral dependency by subcells for different airmasses. I'd be interested in this approach. Has the code been published open access somewhere and has it been included into the pvlib yet?

mikofski, 2019-03-14:
If you know the quantum efficiency of each junction then you can determine the spectral mismatch by integrated over wavelength
[image]
and then comparing it to an equivalent short circuit current in [AM1.5](https://rredc.nrel.gov/solar//spectra/am1.5/).

Maybe this [jupyter notebook](https://nbviewer.jupyter.org/gist/mikofski/54aac3ee295bab8118df) might help? Also you can take a look at my scripts from pvsc44, specifically:
* [`spectral_analysis.py`](https://gitlab.com/bwanamarko/spectralmismatch/blob/master/spectral_analysis.py)
* [`spectral_correction.py`](https://gitlab.com/bwanamarko/spectralmismatch/blob/master/spectral_correction.py)
* [`spectral_shift.py`](https://gitlab.com/bwanamarko/spectralmismatch/blob/master/spectral_shift.py)
* [`spectral_mismatch.py`](https://gitlab.com/bwanamarko/spectralmismatch/blob/master/spectral_mismatch.py)

then add the currents of the IV curves to get the total for the stack.

cwhanse, 2019-03-14:
We developed a piecewise model for the spectral response of a Semprius module (i.e., f(AM) = f1(AM) for AM< threshold, =f2(AM) for AM>=threshold).  I'll have to find the work and see if I can share it.  The result is not in PVLib for Matlab nor in pvlib-python.

cwhanse, 2019-03-14:
[Performance model for Semprius CPV](https://energy.sandia.gov/wp-content//gallery/uploads/Semprius_perf_model_final.pdf)  is this what you are looking for?

Piranias, 2019-03-20:
@cwhanse : yes, that's what I was referring to. My question was more if this approach in already integrated into pvlib (as you said it's not) or if that wouldn't be a good enhancement of the current flat plate module

cwhanse, 2019-03-20:
I haven't considered adding it because it hasn't been requested, and we only have parameters for that Semprius module.  If you build out functions for the model that fit within pvlib I'm interested to see them.

mikofski, 2019-03-23:
@cwhanse and @wholmgren, I realize now that [my comment](#issuecomment-472976951) was irrelevant to this issue,  but it got me thinking that the effect of spectra and quantum efficiency is not documented in pvlib. Do you think it's worth adding?

wholmgren, 2019-03-23:
@mikofski yes.

cwhanse, 2019-03-25:
@mikofski what do you propose?  The pvlib readthedocs doesn't walk through the modeling steps in an orderly way, which I think would be a big help to many users.  We could model it on [www.pvpmc.org](https://pvpmc.sandia.gov/modeling-steps/) and add pages as topics come up, such as spectral effect modeling.