pvlib/pvlib-python #679: Add function for interpolated IAM model

cwhanse, 2019-03-28:
**Describe the solution you'd like**

A commonly used model for IAM is to interpolate along a user-supplied list of (AOI, IAM) points, e.g., pvsyst offers this model option.

Implement a function `iam_interpolated` with parameters being a list of (AOI, IAM) points.

**Additional context**
pvsyst uses linear interpolation.

wholmgren, 2019-03-28:
I'm not entirely opposed to adding a one-line wrapper around numpy or scipy interpolate 1D functions. An alternative to consider is only a documentation example that demonstrates using those functions for this purpose. Or ask and answer the question on stackoverflow.

cwhanse, 2019-03-28:
Good point.  I think it would be useful as a PVSystem method, when the (AOI, IAM) points are packaged with module_parameters.

mikofski, 2019-03-28:
Or perhaps in the model chain? Is there a pvsyst model chain for user supplied IAM lookup?

cwhanse, 2019-07-26:
closed by #752

closed 2019-07-26 by cwhanse