pvlib/pvlib-python #680: Naming pattern for IAM functions

cwhanse, 2019-03-28:
**Is your feature request related to a problem? Please describe.**
It may be easier for users to find and select IAM functions if the function names start with the prefix `iam`, e.g., `iam_ashrae` rather than the current pattern placing `iam` as a suffix `ashraeiam`

**Describe the solution you'd like**
Rename `physicaliam`, `ashraeiam`, etc. and deprecate old names.

wholmgren, 2019-03-28:
Related to https://github.com/pvlib/pvlib-python/issues/436... I currently prefer to see a new module `iam.py` with functions `physical`, `ashrae`, etc.

cwhanse, 2019-10-02:
Starting on a PR for this.  `iam.py` conflicts with the often-used variable `iam`, e.g., `iam = pvsystem.ashraeiam()` in the current code.  Suggestions? Seems like a good idea to keep `iam` available for users to name the output of these functions.

wholmgren, 2019-10-02:
I don't have a better idea, but I don't think we need to worry about what variables might exist in user code. There's only a name conflict when the module is imported as `from pvlib import iam`. While that's a common pattern, I don't think it's frequent enough to justify choosing a worse name for our module.

PR #783 (2019-10-18): Create iam.py, consistent naming for IAM functions
 - [x] Closes  #680. Related to #436 
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style  [... cut]
closed 2019-10-18 by wholmgren