pvlib/pvlib-python #690: Read The Docs not rendering API arguments : types correctly

mikofski, 2019-04-22:
**Describe the bug**
there is no colon between args and types anymore

**Screenshots**
https://pvlib-python.readthedocs.io/en/latest/generated/pvlib.location.Location.get_solarposition.html#pvlib.location.Location.get_solarposition
[image]

**Expected behavior**
there should be colons, like this
https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.location.Location.get_solarposition.html
[image]

wholmgren, 2019-04-22:
The "expected behavior" also looks bad because there's no space between the colons and the types. I'm guessing this is due to numpydoc or the sphinxrtd theme (or both). I saw there's a new numpydoc release so I manually triggered the doc build. Unfortunately there now an error that prevents the build from completing. See here: https://readthedocs.org/projects/pvlib-python/builds/8948978/

mikofski, 2019-04-22:
From the error I think it's because the ["see also" items in `solar_zenith_analytical`](../blob/ef82a39/pvlib/solarposition.py#L1300) are all on the same line without commas. I think I should have put them on separate lines.

mikofski, 2019-04-23:
Okay, I think I got it working, needed `sphinx.ext.napoleon` extension, see [this blogger](https://samnicholls.net/2016/06/15/how-to-sphinx-readthedocs/)

[image]

much better, right? there are a couple of options you can read about them here in [napoleon docs](http://www.sphinx-doc.org/en/master/usage/extensions/napoleon.html#configuration), numpydoc is enabled by default. I played around with them, but I couldn't really see any differences.

wholmgren, 2019-04-24:
The "latest" docs are building successfully, so I'll close this.

[image]

closed 2019-04-24 by wholmgren