pvlib/pvlib-python #695: Temperature dependence of Isc not included in sapm model for some parameters

toddkarin, 2019-04-30:
The temperature dependence of the effective irradiance is:

Ee = Isc / [Isco⋅{1+αIsc⋅(Tc-To)}]

as given in the Sandia model [King2004]. The function pvsystem.sapm() has the following line for calculating Isc:
```
out['i_sc'] = (module['Isco'] * Ee * (1 + module['Aisc']*(temp_cell - T0)))
 ```
which correctly takes into account the temperature dependence of Isc.

However, for calculating Voc, the relevant lines in pvsystem.sapm are:
```
np.log(Ee, where=Ee_gt_0, out=logEe)
...
out['v_oc'] = np.maximum(0, (
        module['Voco'] + cells_in_series * delta * logEe +
        Bvoco*(temp_cell - T0)))
```
Here we note that the temperature coefficient of the short circuit current is not used when calculating Voc (and also a few other parameters). 

One way to fix this would be to replace the line for calculating log effective irradiance with:
```
np.log(Ee* (1 + module['Aisc']*(temp_cell - T0)), where=Ee_gt_0, out=logEe)
```

**References**

[King2004] D.L. King, W.E. Boyson, J.A. Kratochvill. Photovoltaic Array Performance Model. SAND2004-3535. 2004

cwhanse, 2019-04-30:
@toddkarin the King 2004 report is really confusing on the calculation sequence, which is different if starting with irradiance as opposed to starting with measured Isc.  If starting with irradiance (the pvlib sequence), effective irradiance Ee is calculated by Eq. 21, and the value Ee is used in Eq. 1 through 4 to calculate Isc, Imp, Voc and Vmp.  Specifically in Eq. 1, Ee replaces the expression f1(AM)*{Eb… / Eo}.

However, when analyzing measured module IV curves, Ee is calculated from Isc using Eq. 7, and the value of Ee is used in Eq. 2 through 4 to model the other quantities.

I hope this explanation is clear.

toddkarin, 2019-04-30:
I've looked into it further, and think that while the equation is not strictly correct, it is an appropriate first order approximation. Take a look at the attached derivation:

[Scan Apr 30, 2019 at 1.00 PM.pdf](https://github.com/pvlib/pvlib-python/files/3132946/Scan.Apr.30.2019.at.1.00.PM.pdf)

The last equation shows that the second term with the alpha_isc is missing from Eq. 3. alpha_isc is typically 0.05% per Kelvin. This means that with T = T0 + 50 C, the second term is (25 mV)*log(1+0.0005*50) = 0.6 mV. This creates an error of 0.1% for a cell with 600 mV Voc. This is not a big problem for reasonable temperatures and small temperature coefficients of Isc.

Another way to treat this missing term is to expand log(1+x)=x and then combine it into the definition of Bvoc. You then end up with an equation for the first order temperature correction of Bvoc:

Bvoc = Bvoco + nk/e*alpha_Isc*(T-T0)

This method demonstrates that the missing term is a second order effect in temperature deviation and thus ignorable.

cwhanse, 2019-05-03:
Thanks for the explanation @toddkarin .  I'm going to close this as an issue for pvlib, because the `pvsystem.sapm` function implements the model in the King 2004 report.  Improvements to model in King 2004 are welcome but I don't think this is the forum for that.

closed 2019-05-03 by cwhanse