pvlib/pvlib-python #696: PVWatts_loss was applied on AC calculation

yxh289, 2019-05-01:
Hi, 
    In the description of pvwatts_losses function, the losses are applied on DC side:
'Calculates DC power losses according the PVwatts model using pvwatts_losses() and self.losses_parameters.`' 
    However I found that in pvlib.modelchain.ModelChain.pvwatts_losses, PVwatts loss was applied to AC. 
'def pvwatts_losses(self):
        self.losses = (100 - self.system.pvwatts_losses()) / 100.
        self.ac *= self.losses
        return self'
   When system have high DC/AC ratio, AC power output clips at a lower level as a result of these losses applied on the AC side.

wholmgren, 2019-05-02:
thanks @yxh289. pull request welcome.

cwhanse, 2019-05-03:
Concur with the bug report.  May have been due to coding following the older PVWatts description, rather than the v5 document.  Or a simple typo.

closed 2019-05-13 by wholmgren

pasquierjb, 2022-09-01:
The post-inverter losses such as the AC wiring losses and AC Availability should in theory only apply to the AC power and not be subjected to inverter clipping.

kanderso-nrel, 2022-09-01:
@pasquierjb conceptually that might be true, but in this case we're just sticking with the original spec: per Section 11 of the [PVWatts Version 5 Manual](https://www.nrel.gov/docs/fy14osti/62641.pdf), "Losses in the system that are not explicitly modeled are provided by the user as a percentage of DC energy."

See also #1350

PR #704 (2019-05-13): change MC to apply pvwatts_losses to mc.dc instead of mc.ac
 - [x] Closes #696
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style is suffici [... cut]