pvlib/pvlib-python #705: Remove clear sky fallback option from ModelChain.prepare_inputs and run_model

wholmgren, 2019-05-03:
`ModelChain.prepare_inputs` and `ModelChain.run_model` currently default to running a model using clear sky data if none of `ghi`, `dni`, and `dhi` are in the input `weather`.

My original motivation for including this feature was that it would make it easier to get started with ModelChain. But it's not practically useful and arguably makes the documentation of an already confusing class even worse. 

It also violates the "explicit is better than implicit" guidance -- better to require that users provide their own data. Their own data is only a method call away with `location.get_clearsky`.

cwhanse, 2019-05-03:
I agree with removing the implicit clearsky weather.

wholmgren, 2019-05-03:
Ok, I'm going to put a deprecation warning into 0.6.2.

wholmgren, 2019-11-04:
closed by #773

closed 2019-11-04 by wholmgren