pvlib/pvlib-python #714: Improve docstring for iotools.tmy.read_tmy3 and read_tmy2

cwhanse, 2019-05-07:
Docstrings are text copied from the original MATLAB functions. Issues include:
- inconsistent with the pvlib variable naming convention, e.g. GHI instead of ghi
- not formatted to render properly in sphinx
- inconsistent with actual function (columns names in returned data are not what is described in the table).

AdamRJensen, 2022-08-01:
Hi @cwhanse - do you know how many of these issues are still relevant (besides the variable mapping which I know is lacking)?

cwhanse, 2022-08-01:
@AdamRJensen I wish I'd listed specifics when I opened this issue. I think 2, probably 3 and 4 are fixed in #1494. 5 is similar to 1, in that `iotools.read_tmy3` returns with column names consistent with TMY3 documentation, but not with pvlib internal variables. In contrast, `iotools.read_epw` uses pvlib-python names where the columns are a pvlib quantity. 

I am not advocating for `read_tmy3` to return pvlib-python names; keeping TMY3 names helps with debugging, maintenance, etc. Maybe a `tmy3_to_pvlib` function that renames or copies certain columns would be appreciated.

AdamRJensen, 2022-08-02:
The `read_tmy3` function certainly shouldn't only return pvlib variable names, but I suppose it is desirable to have the `map_variables` options as many of the other iotools? Ideally, I think all the iotools functions should have `map_variables=True` by default.

cwhanse, 2022-08-02:
Oh yes I had forgotten about the `map_variables` capability. I agree, it would be good to add to `read_tmy3`. In a new PR, though.

I wonder what use is being made of the current `rename_columns`. The new names are more readable. Are they also internal names for some other software?

wholmgren, 2022-08-02:
> Are they also internal names for some other software?

as you may know, they match MATLAB PVLIB's [tmy reader](https://github.com/sandialabs/MATLAB_PV_LIB/blob/master/pvl_readtmy3.m). I'm not aware of others.

AdamRJensen, 2022-08-02:
I would suggest we deprecate the `recolumn` argument (which removes the units from the column names) and instead add `map_variables` (which would map the original column names to pvlib names). Thoughts?

    recolumn (bool, default True) – If True, apply standard names to 
    TMY3 columns. Typically this results in stripping the units from the column name.

The description also seems misleading, as `recolumn` does not apply the pvlib "standard names"..

AdamRJensen, 2022-08-05:
@cwhanse @wholmgren Can I go ahead and close this issue and open a new issue/PR concerning the deprecation of `rename_columns` and addition of `map_variables` to the `read_tmy3` function?

closed 2022-08-08 by AdamRJensen